slic3r/Slic3r #3929: Bad bridging

swiftgeek, 2017-05-03:
### Version
8c77f476
### Operating system type + version
Archlinux
### Behavior
[image]
[image]
Bridge happens along longer distance instead of short one

#### STL/Config (.ZIP) where problem occurs
[bad-bridging-8c77f476.zip](https://github.com/alexrj/Slic3r/files/973377/bad-bridging-8c77f476.zip)

swiftgeek, 2017-05-03:
Another example
[image]
[bridge-to-air.zip](https://github.com/alexrj/Slic3r/files/974306/bridge-to-air.zip)

lordofhyphens, 2017-05-04:
Is this a recent development (aka regression)?

alexrj, 2017-05-10:
Simplified test case: [x-end_trim.stl.zip](https://github.com/alexrj/Slic3r/files/990594/x-end_trim.stl.zip)

alexrj, 2017-05-10:
The problem is caused by this irregular shape in object:
[image]

Here's a screenshot of the anchor layer (the one below the bridge). The grey shape is the original sliced model. Notice the white gap.
[image]

alexrj, 2017-05-10:
Fixed. Bridge detection will now deal with situations like that where lower layer has narrow gaps under the bridge anchors.

closed 2017-05-10 by alexrj