slic3r/Slic3r #3935: group of layer parts omitted from gcode

jluscher, 2017-05-05:
### Version
1.2.9

### Operating system type + version
Ubuntu 17.04

### Behavior
Layers are partially missing when sliced - beetween a pair of holes but not another(?!).

Preview:
[image]

Close up of preview:
[image]

The imported .stl file looks good:
[image]

Testing in Cura (for comparison) showed no problem - the preview shows the missing portion is present there:
[image]

Printed parts from slic3r:
[image]

#### STL/Config (.ZIP) where problem occurs
[slic3r_config-and-stl.zip](https://github.com/alexrj/Slic3r/files/979582/slic3r_config-and-stl.zip)

xoan, 2017-05-05:
Your model has some errors inside the top drill and it is non-manifold.

[image]

alexrj, 2017-05-05:
You might also want to try with 1.3.0-dev

lordofhyphens, 2017-05-12:
Recent build hard-crashes instead on load. 

Ran the model through netfabb online repair, and all is well.
[lamp-bracket_v25_fixed.zip](https://github.com/alexrj/Slic3r/files/994746/lamp-bracket_v25_fixed.zip)

closed 2017-05-12 by lordofhyphens