slic3r/Slic3r #3937: Current development version does not start, missing library

simonCor, 2017-05-07:
### Version
slic3r-1.3.0-dev-1d97698-linux-x64

### Operating system type + version
Linux, openSuse 42.2 (42.2.20161212)

### Behavior
When I start Slicer I get an error about a missing library. The stable 1.2.9 works.

```
./Slic3r --gui
Can't load '/home/sschwarz/Downloads/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Wx/Wx.so' for module Wx: libwx_gtk2u_adv-3.0.so.0: cannot open shared object file: No such file or directory at DynaLoader.pm line 193.
 at /home/sschwarz/Downloads/Slic3r/local-lib/lib/perl5/Slic3r/GUI.pm line 7.
Compilation failed in require at /home/sschwarz/Downloads/Slic3r/local-lib/lib/perl5/Slic3r/GUI.pm line 7.
BEGIN failed--compilation aborted at /home/sschwarz/Downloads/Slic3r/local-lib/lib/perl5/Slic3r/GUI.pm line 7.
Compilation failed in require at (eval 84) line 1.
```

Tinchus2009, 2017-05-09:
Check you local installation again. I just builted latest dev version, under win 10 and it works ok

lordofhyphens, 2017-05-09:
@Tinchus2009 he specifically said linux, for which I know is experimental
and am working out a solution.

lordofhyphens, 2017-05-21:
@simonCor can you give a recent build a try? I can't reproduce this on my own systems. It's probably a library naming issue though (the build was done on Ubuntu 14.04)

lordofhyphens, 2017-05-21:
`ldd /home/sschwarz/Downloads/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Wx/Wx.so' for module Wx: libwx_gtk2u_adv-3.0.so.0` will tell you what library is "missing", but please try a recent build from dl.slic3r.org/dev/linux

thuSkywalker, 2017-05-21:
I'm receiving following error when running `./Slic3r --gui`:

```
./perl-local: relocation error: ./bin/libm.so.6: symbol __get_cpu_features, version GLIBC_PRIVATE not defined in file libc.so.6 with link time reference
```

**Version:** slic3r-1.3.0-dev-450a918-linux-x64

lordofhyphens, 2017-05-21:
@thuSkywalker That isn't strictly related to this problem, please open a separate issue.

simonCor, 2017-05-22:
@lordofhyphens I tried it with a recent build (slic3r-1.3.0-dev-450a918-linux-x64). It gives the same result.

Executing ldd give the following:
`ldd /home/sschwarz/Downloads/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Wx/Wx.so
	linux-vdso.so.1 (0x00007fff3dbc5000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007fe7e034c000)
	libwx_gtk2u_adv-3.0.so.0 => not found
	libwx_gtk2u_core-3.0.so.0 => not found
	libwx_baseu-3.0.so.0 => not found
	libstdc++.so.6 => /usr/lib64/libstdc++.so.6 (0x00007fe7dffc1000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007fe7dfdaa000)
	libc.so.6 => /lib64/libc.so.6 (0x00007fe7dfa07000)
	/lib64/ld-linux-x86-64.so.2 (0x000055f10016b000)
	libm.so.6 => /lib64/libm.so.6 (0x00007fe7df709000)`

Searching for the missing libwx_gtk2u_adv in the package manager gives clue i thing:
`zypper search libwx_gtk2u_adv
Loading repository data...
Reading installed packages...

S | Name                                            | Summary          | Type   
--+-------------------------------------------------+------------------+--------
  | libwx_gtk2u_adv-2_8-0-compat-lib-stl            | wxWidgets Libr-> | package
  | libwx_gtk2u_adv-2_8-0-compat-lib-wxcontainer    | wxWidgets Libr-> | package
  | libwx_gtk2u_adv-2_8-0-compat-lib-wxcontainer24c | wxWidgets Libr-> | package
  | libwx_gtk2u_adv-2_8-0-stl                       | wxWidgets Libr-> | package
  | libwx_gtk2u_adv-2_8-0-stl-32bit                 | wxWidgets Libr-> | package
  | libwx_gtk2u_adv-2_8-0-wxcontainer               | wxWidgets Libr-> | package
  | libwx_gtk2u_adv-2_8-0-wxcontainer24c            | wxWidgets Libr-> | package
  | libwx_gtk2u_adv-suse-nostl1                     | wxWidgets Libr-> | package
i | libwx_gtk2u_adv-suse-nostl3                     | wxWidgets Libr-> | package
  | libwx_gtk2u_adv-suse-nostl3-debuginfo           | Debug informat-> | package
i | libwx_gtk2u_adv-suse1                           | wxWidgets Libr-> | package
  | libwx_gtk2u_adv-suse1-32bit                     | wxWidgets Libr-> | package`

It seems that openSuse leap 42.2 does not ship with the versision 3.0 yet.

lordofhyphens, 2017-05-23:
So us shipping wxwidgets at this point is going to cause other problems. So
that is a no-go, you will probably have to find an opensuse package for
wxwidgets 3.0 or build yourself.

Even debian stable has 3.0 in packages.

When we go full c++ I will probably ship Slic3r statically linked with most
of its dependencies or try to include a build of wxwidgets that has its own
deps statically included in itself.

simonCor, 2017-05-24:
@lordofhyphens seems I was a bit to fast with the wxWidgets version.

```
zypper info libwx_gtk2u_adv-suse1
Loading repository data...
Reading installed packages...

Information for package libwx_gtk2u_adv-suse1:
----------------------------------------------
Repository     : openSUSE-42.2-0             
Name           : libwx_gtk2u_adv-suse1       
Version        : 3.0.2-7.18                  
Arch           : x86_64                      
Vendor         : openSUSE                    
Installed Size : 1.8 MiB                     
Installed      : Yes                         
Status         : up-to-date                  
Source package : wxWidgets-3_0-3.0.2-7.18.src
Summary        : wxWidgets Library           
Description    :                             
    Library for the wxWidgets cross-platform GUI
```
So from the package system the library is installed. But Slicer does not find it.

xoan, 2017-05-31:
Similar issue, different library:

```
% ./Slic3r --gui
Can't load '/home/xoan/Descargas/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Wx/Wx.so' for module Wx: libjbig.so.0: cannot open shared object file: No such file or directory at DynaLoader.pm line 193.
 at /home/xoan/Descargas/Slic3r/local-lib/lib/perl5/Slic3r/GUI.pm line 7.
Compilation failed in require at /home/xoan/Descargas/Slic3r/local-lib/lib/perl5/Slic3r/GUI.pm line 7.
BEGIN failed--compilation aborted at /home/xoan/Descargas/Slic3r/local-lib/lib/perl5/Slic3r/GUI.pm line 7.
Compilation failed in require at (eval 84) line 1.
```

```
% ldd /home/xoan/Descargas/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Wx/Wx.so
	linux-vdso.so.1 (0x00007ffe57848000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007feea3ed8000)
	libwx_gtk2u_adv-3.0.so.0 => /usr/lib/libwx_gtk2u_adv-3.0.so.0 (0x00007feea3af6000)
	libwx_gtk2u_core-3.0.so.0 => /usr/lib/libwx_gtk2u_core-3.0.so.0 (0x00007feea3277000)
	libwx_baseu-3.0.so.0 => /usr/lib/libwx_baseu-3.0.so.0 (0x00007feea2de1000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007feea2a58000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007feea2841000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007feea249a000)
	/usr/lib64/ld-linux-x86-64.so.2 (0x00005564eb09e000)
	libgtk-x11-2.0.so.0 => /usr/lib/libgtk-x11-2.0.so.0 (0x00007feea1e5a000)
	libgdk-x11-2.0.so.0 => /usr/lib/libgdk-x11-2.0.so.0 (0x00007feea1ba5000)
	libgdk_pixbuf-2.0.so.0 => /usr/lib/libgdk_pixbuf-2.0.so.0 (0x00007feea197e000)
	libpango-1.0.so.0 => /usr/lib/libpango-1.0.so.0 (0x00007feea1731000)
	libgobject-2.0.so.0 => /usr/lib/libgobject-2.0.so.0 (0x00007feea14df000)
	libglib-2.0.so.0 => /usr/lib/libglib-2.0.so.0 (0x00007feea11c9000)
	libX11.so.6 => /usr/lib/libX11.so.6 (0x00007feea0e8a000)
	libpangocairo-1.0.so.0 => /usr/lib/libpangocairo-1.0.so.0 (0x00007feea0c7c000)
	libcairo.so.2 => /usr/lib/libcairo.so.2 (0x00007feea094e000)
	libXxf86vm.so.1 => /usr/lib/libXxf86vm.so.1 (0x00007feea0748000)
	libSM.so.6 => /usr/lib/libSM.so.6 (0x00007feea0540000)
	libpng16.so.16 => /usr/lib/libpng16.so.16 (0x00007feea0308000)
	libjpeg.so.8 => /usr/lib/libjpeg.so.8 (0x00007feea009e000)
	libtiff.so.5 => /usr/lib/libtiff.so.5 (0x00007fee9fe18000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007fee9fb06000)
	libz.so.1 => /usr/lib/libz.so.1 (0x00007fee9f8ef000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x00007fee9f6eb000)
	libgmodule-2.0.so.0 => /usr/lib/libgmodule-2.0.so.0 (0x00007fee9f4e5000)
	libXfixes.so.3 => /usr/lib/libXfixes.so.3 (0x00007fee9f2df000)
	libatk-1.0.so.0 => /usr/lib/libatk-1.0.so.0 (0x00007fee9f0b9000)
	libgio-2.0.so.0 => /usr/lib/libgio-2.0.so.0 (0x00007fee9ed24000)
	libpangoft2-1.0.so.0 => /usr/lib/libpangoft2-1.0.so.0 (0x00007fee9eb0e000)
	libfontconfig.so.1 => /usr/lib/libfontconfig.so.1 (0x00007fee9e8ca000)
	libXrender.so.1 => /usr/lib/libXrender.so.1 (0x00007fee9e6bd000)
	libXinerama.so.1 => /usr/lib/libXinerama.so.1 (0x00007fee9e4ba000)
	libXi.so.6 => /usr/lib/libXi.so.6 (0x00007fee9e2a9000)
	libXrandr.so.2 => /usr/lib/libXrandr.so.2 (0x00007fee9e09e000)
	libXcursor.so.1 => /usr/lib/libXcursor.so.1 (0x00007fee9de93000)
	libXcomposite.so.1 => /usr/lib/libXcomposite.so.1 (0x00007fee9dc90000)
	libXdamage.so.1 => /usr/lib/libXdamage.so.1 (0x00007fee9da8b000)
	libXext.so.6 => /usr/lib/libXext.so.6 (0x00007fee9d879000)
	libthai.so.0 => /usr/lib/libthai.so.0 (0x00007fee9d66f000)
	libffi.so.6 => /usr/lib/libffi.so.6 (0x00007fee9d466000)
	libpcre.so.1 => /usr/lib/libpcre.so.1 (0x00007fee9d1f3000)
	libxcb.so.1 => /usr/lib/libxcb.so.1 (0x00007fee9cfc8000)
	libfreetype.so.6 => /usr/lib/libfreetype.so.6 (0x00007fee9cd01000)
	libpixman-1.so.0 => /usr/lib/libpixman-1.so.0 (0x00007fee9ca59000)
	libEGL.so.1 => /usr/lib/libEGL.so.1 (0x00007fee9c846000)
	libxcb-shm.so.0 => /usr/lib/libxcb-shm.so.0 (0x00007fee9c642000)
	libxcb-render.so.0 => /usr/lib/libxcb-render.so.0 (0x00007fee9c434000)
	libGL.so.1 => /usr/lib/libGL.so.1 (0x00007fee9c1a7000)
	librt.so.1 => /usr/lib/librt.so.1 (0x00007fee9bf9f000)
	libICE.so.6 => /usr/lib/libICE.so.6 (0x00007fee9bd82000)
	libuuid.so.1 => /usr/lib/libuuid.so.1 (0x00007fee9bb7d000)
	liblzma.so.5 => /usr/lib/liblzma.so.5 (0x00007fee9b957000)
	libresolv.so.2 => /usr/lib/libresolv.so.2 (0x00007fee9b73f000)
	libmount.so.1 => /usr/lib/libmount.so.1 (0x00007fee9b4f1000)
	libharfbuzz.so.0 => /usr/lib/libharfbuzz.so.0 (0x00007fee9b25d000)
	libexpat.so.1 => /usr/lib/libexpat.so.1 (0x00007fee9b033000)
	libdatrie.so.1 => /usr/lib/libdatrie.so.1 (0x00007fee9ae2b000)
	libXau.so.6 => /usr/lib/libXau.so.6 (0x00007fee9ac25000)
	libXdmcp.so.6 => /usr/lib/libXdmcp.so.6 (0x00007fee9aa1f000)
	libbz2.so.1.0 => /usr/lib/libbz2.so.1.0 (0x00007fee9a80f000)
	libGLdispatch.so.0 => /usr/lib/libGLdispatch.so.0 (0x00007fee9a559000)
	libGLX.so.0 => /usr/lib/libGLX.so.0 (0x00007fee9a327000)
	libblkid.so.1 => /usr/lib/libblkid.so.1 (0x00007fee9a0df000)
	libgraphite2.so.3 => /usr/lib/libgraphite2.so.3 (0x00007fee99eb3000)
```

lordofhyphens, 2017-05-31:
Distro?

xoan, 2017-05-31:
Ups, sorry: `Linux 4.9.30-1-lts x86_64 GNU/Linux`

lordofhyphens, 2017-05-31:
Is libjpeg installed from sources?

lordofhyphens, 2017-05-31:
Sorry, packages.

xoan, 2017-05-31:
Yep, from `extra` repository.

```
% pacman -Qs libjpeg        
local/libjpeg-turbo 1.5.1-1
    JPEG image codec with accelerated baseline compression and decompression
```

https://www.archlinux.org/packages/extra/x86_64/libjpeg-turbo/

lordofhyphens, 2017-05-31:
As a workaround, add a symlink in the dir with the other libraries with the
name libjpeg.so.0 to /usr/lib/libjpeg.so.8

It's a distro difference.

xoan, 2017-05-31:
It is not `libjpeg` but `libjbig`.

Arch Linux has no `/usr/lib/libjbig.so.0` in any related package: https://www.archlinux.org/packages/?q=jbig

lordofhyphens, 2017-05-31:
https://packages.ubuntu.com/trusty/libjbig-dev

If you installed libwxgtk and removed the related files from  the library
dir in Slic3r it would probably work.

lordofhyphens, 2017-05-31:
I'll add that back to the list of files to have in the package (it is apparently esoteric).

If the wxperl libs weren't picky about what they linked to I'd just remove the regular wxwidgets libs and require that the user installs them.

lordofhyphens, 2017-06-03:
Okay, libjbig is in the package now. Hopefully something else doesn't break -.-;

closed 2017-06-03 by lordofhyphens