slic3r/Slic3r #3938: Unable to run Linux dev builds

mrvanes, 2017-05-07:
### Version
1.3.0-dev-x64 test build

### Operating system type + version
x64 Linux Ubuntu Zesty

### Behavior
I download latest Linux dev build from http://dl.slic3r.org/dev/linux/, unpack and run ./Slic3r --gui

Result:
```
$ ./Slic3r --gui
Can't load '/home/martin/Downloads/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Wx/Wx.so' for module Wx: /home/martin/Downloads/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Wx/Wx.so: symbol _ZThn704_N17wxGenericListCtrl31GetSizeAvailableForScrollTargetERK6wxSize, version WXU_3.0 not defined in file libwx_gtk2u_core-3.0.so.0 with link time reference at DynaLoader.pm line 193.
 at /home/martin/Downloads/Slic3r/local-lib/lib/perl5/Slic3r/GUI.pm line 7.
Compilation failed in require at /home/martin/Downloads/Slic3r/local-lib/lib/perl5/Slic3r/GUI.pm line 7.
BEGIN failed--compilation aborted at /home/martin/Downloads/Slic3r/local-lib/lib/perl5/Slic3r/GUI.pm line 7.
Compilation failed in require at (eval 84) line 1.
```
I've tried a couple, none of them will compile.

curieos, 2017-05-07:
I think the --gui flag is to check that you have all the dependencies installed (more for a source build than a precompile). Did you try running it without the --gui flag?

curieos, 2017-05-07:
also this looks like a duplicate of #3937, are you on the same version?

mrvanes, 2017-05-07:
Running Slic3r without --gui gives me the help output of Slic3r and no gui. It's not a duplicate of #3937 since that mentions a missing library while for me there's an unresolved library symbol/version issue.

mrvanes, 2017-05-08:
After some testing it turns out the -dev build is using my pre-installed OS version of libwx_gtk2u_core-3.0.so.0.2.0. I'm using the Slicer batch file and expect it to use the local perl environment but that apparently fails. Am I missing something here?

mrvanes, 2017-05-08:
It helps when I add the local bin/ directory to LD_LIBRARY_PATH. But then, local libm.so.6 can't locate __get_cpu_features in system libc.so.6 which seems to be [relocated to ld-linux-x86-64.so.2 since Ubuntu 16.04](https://bugs.launchpad.net/ubuntu/+source/glibc/+bug/1572569).
I may be completely off here, but it looks like a discrepancy between the travis build environment and my Zesty laptop?

lordofhyphens, 2017-05-08:
The precompiled builds uses a built Perl that it comes with, not system
perl.

mrvanes, 2017-05-08:
I know, that's the problem... the included bash file wrongly points LD_LIBRARY_PATH to ./shlib/x86_64-linux-thread-multi/ (there is no top-level shlib dir) so Wx.so fails initially and when corrected to ./bin the included .so libraries (libm.so.6 and libresolv.so.2) have a broken dependancy on *system* libc.so.6.
I have used the included perl-local binary all the time.

lordofhyphens, 2017-05-08:
Yeah, I feel as if I need to grab every stupid library down the line. Fun
thing is that this was packaged on Ubuntu 14.04 LTS

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

mrvanes, 2017-05-08:
So, long story short, for anyone reading along and wanting to get this working on Ubuntu Zesty (17.04).
Change the export LD_LIBRARY_PATH line in Slic3r script to

```export LD_LIBRARY_PATH=./bin```

and remove libm.so.6 and libresolv.so.2 from /bin directory. Then run

```./Slic3r --gui```

and all is good.

lordofhyphens, 2017-05-21:
I've fixed the startup script to export LD_LIBRARY_PATH to the correct dir; I am going to try removing libm.so.6 and libresolv.so.2 and see what other distros it breaks.

TheCuriousForge, 2017-05-21:
I am running Ubuntu 14.04. I just downloaded the latest beta and am unable to start Silc3r. Here's the output:

Can't load '/home/curiousforge/3D/Slic3r beta/beta/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so' for module Slic3r::XS: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by /home/curiousforge/3D/Slic3r beta/beta/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so) at DynaLoader.pm line 193.
 at /home/curiousforge/3D/Slic3r beta/beta/Slic3r/local-lib/lib/perl5/Slic3r.pm line 53.
Compilation failed in require at /home/curiousforge/3D/Slic3r beta/beta/Slic3r/local-lib/lib/perl5/Slic3r.pm line 53.
BEGIN failed--compilation aborted at /home/curiousforge/3D/Slic3r beta/beta/Slic3r/local-lib/lib/perl5/Slic3r.pm line 53.
Compilation failed in require at ./slic3r.pl line 16.
BEGIN failed--compilation aborted at ./slic3r.pl line 16

TheCuriousForge, 2017-05-21:
looks like this is the problem:
 /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.8'

lordofhyphens, 2017-05-21:
Yeah, that makes no sense to me as these are built on Ubuntu 14.04.
However, we did install gcc 4.9 from distro which may have upgraded the
ABI. In which case, I can't do anything about that.

lordofhyphens, 2017-05-21:
We could include our libstdc++ but I am a bit leery about compatibility
with other distros.

Pain, pain, pain.

TheCuriousForge, 2017-05-21:
I updated gcc to 4.9 and it works.

mrvanes, 2017-05-22:
To add some oil to the fire; I only need Slic3r's libwx_* libraries in ./bin to make it work. I can (and did) delete the rest and use Zesty system libs instead.

lordofhyphens, 2017-05-22:
That is because you have everything else from packages.

theopensourcerer, 2017-05-22:
I'm on Ubuntu 16.10 and can't run yesterday's latest dev build (450a918) either:

`alanlord@lobsang:~$ bin/Slic3r/Slic3r --gui
Can't load '/home/alanlord/bin/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Wx/Wx.so' for module Wx: /home/alanlord/bin/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Wx/Wx.so: symbol _ZThn704_N17wxGenericListCtrl31GetSizeAvailableForScrollTargetERK6wxSize, version WXU_3.0 not defined in file libwx_gtk2u_core-3.0.so.0 with link time reference at DynaLoader.pm line 193.
 at /home/alanlord/bin/Slic3r/local-lib/lib/perl5/Slic3r/GUI.pm line 7.
Compilation failed in require at /home/alanlord/bin/Slic3r/local-lib/lib/perl5/Slic3r/GUI.pm line 7.
BEGIN failed--compilation aborted at /home/alanlord/bin/Slic3r/local-lib/lib/perl5/Slic3r/GUI.pm line 7.`

mrvanes, 2017-05-23:
@lordofhyphens Yes, indeed, but I have libwx also from packages, but that's where the hurt starts.

lordofhyphens, 2017-06-03:
@mrvanes http://dl.slic3r.org/dev/linux/slic3r-1.3.0-dev-bcd8603-linux-x64.tar.bz2 ?

mrvanes, 2017-06-05:
Runs like a charm, no changes necessary anymore! Tested with dev-9f0335b.

lordofhyphens, 2017-06-05:
I think this is "good enough" to close, will deal with other problems as they crop up.

closed 2017-06-05 by lordofhyphens