slic3r/Slic3r #3939: Please put comment block at beginning of g-code

jltx1, 2017-05-07:
### Version
1.2.9

### Operating system type + version
Mac OS 10.12.4

### Behavior
Please move the settings comment block in the g-code from the end to the beginning since I need to pull params from there for post-processing scripts which are required to run inline '-i'.   

For example, manipulating layers I need to know layer_height and retract_lift to help differentiate, and raft_layers to skip.   There are many more examples.  I hope you appreciate.  Thanks.

curieos, 2017-05-07:
I don't think that is going to happen because of issue #3849.

LoH was talking about backwards compatibility for this feature so moving it to the front would tend to break this functionality.

jltx1, 2017-05-07:
Good find.  My request would align with his request as well, except for the backwards compatible part.  However, It would be trivial to have an offline script move the settings block to the front for already exported g-code.  My problem is that I must run inline which prevents using such a script.  I am happy to supply such a script for people wanting to use LoH's new feature with older g-code.

alexrj, 2017-05-08:
@jltx1, you inverted the problem with your proposed solution. Let's start from the problem.
As far as I understand from your report, your problem is: I need to access config options from my post-processing script.

The solution is: you can access them as environment variables (`SLIC3R_LAYER_HEIGHT` etc.). This is [http://manual.slic3r.org/advanced/post-processing](documented in the manual).

jltx1, 2017-05-09:
Apologies.  I did RTFM at one point.  :-)

closed 2017-05-09 by jltx1