slic3r/Slic3r #3944: Automated dev build 1c4b860 does not start

cerdeira, 2017-05-10:
Version

Auto dev 1c4b860

Operating system type + version

Win10 64 home

Behavior

Program does not start

lordofhyphens, 2017-05-11:
It's a test failure.

Still working on making Appveyor fail the build when this happens.
````
#   Failed test 'path length is greater than straight line'
#   at t\18_motionplanner.t line 39.
#   Failed test 'path is fully contained in expolygon'
#   at t\18_motionplanner.t line 42.
# Looks like you failed 2 tests of 20.
t\18_motionplanner.t ............ 
````

lordofhyphens, 2017-05-11:
Test failure is still there @alexrj but when the build server changes are merged in it will should properly fail the build on a test failure.

lordofhyphens, 2017-05-11:
@cerdeira FYI, running `slic3r-debug-console.exe` will give more diagnostic information.

lordofhyphens, 2017-05-11:
@cerdeira slic3r-master.2017.05.10.1006.b12453c.zip is the newest build that will work. I don't currently have access to the keys to do so, but I should get it done later (or @alexrj can delete them).

cerdeira, 2017-05-11:
slic3r-debug-console.exe does not start either.

lordofhyphens, 2017-05-11:
Sorry you need to run it from a command line to see the message.

cerdeira, 2017-05-12:
Right,

C:\Ze\3D\Slic3r>slic3r-debug-console.exe
Can't locate Slic3r/XS.pm in @INC (you may need to install the Slic3r::XS module) (@INC contains: C:\Ze\3D\Slic3r\local-lib\lib\perl5\5.24.1\MSWin32-x64-multi-thread C:\Ze\3D\Slic3r\local-lib\lib\perl5\5.24.1 C:\Ze\3D\Slic3r\local-lib\lib\perl5\MSWin32-x64-multi-thread C:\Ze\3D\Slic3r\local-lib\lib\perl5 C:/Ze/3D/Slic3r/lib .) at C:/Ze/3D/Slic3r/lib/Slic3r.pm line 53.
BEGIN failed--compilation aborted at C:/Ze/3D/Slic3r/lib/Slic3r.pm line 53.
Compilation failed in require at C:\Ze\3D\Slic3r\slic3r.pl line 16.
BEGIN failed--compilation aborted at C:\Ze\3D\Slic3r\slic3r.pl line 16.

alexrj, 2017-05-12:
I pushed a fix so the latest build should work.

closed 2017-05-12 by alexrj