slic3r/Slic3r #3945: Override-Options get overridden after closing any settings dialog

foreachthing, 2017-05-11:
### Version
slic3r-master.2017.05.02.995.8c77f47

### Operating system type + version
Windows 7 Pro

### Behavior
#### current
I have a bunch of Print Settings set as _overrides_. But if I open/close (no changes made!) another settings dialog (Printer Settings, Filament or Printer) the already overridden values get over-overridden with the according defaults from the opened/closed dialog!

#### expected
Don't change nothing! Please, please leave my override-settings alone -- thank you!

Or, what are the overrides for, if they get _overridden_?

### Example:
Set _Layer Height_ as override. Set a default value in _Print Settings_ (i.e. 0.2). Then [image] this option to the main panel as override. Reset the option to 0.1. Then go to _Printer_ settings change something (or not) and close the dialog. Notice, that your previously set value of 0.1 gets changed to 0.2!

#### STL/Config (.ZIP) where problem occurs
any.

VanessaE, 2017-05-11:
Fwiw, if you open and then close a settings dialog with no changes, the model gets marked as needs-resliced, as well.

gege2b, 2017-05-12:
previsible, I think it's a 50-pros/50-cons bug/feature :)

#3814 

Keeping your example, you have layer height as override, you open the profile dialog to change some settings related to layer generation (say, # of top solid layers)

at this time, you decide that layer height needs to be changed too (for whatever reason), so you change it here. That's the way we always dealed with settings before overrides.

With the "original" behaviour, when you close the settings dialog, you'll end up with all the settings OK except the layer height, because override.
It's clearly visible with the "generate support" option, but less with layer height so it could waste a print

It's - IMHO - not ergonomic because we are already used to test settings this way

@alexrj tried something in the gui3-shortcuts branch ( https://github.com/alexrj/Slic3r/tree/gui3-shortcuts ) where overrides are replaced with shortcuts instead (it's not so different, they are fully synced with the profile settings, and dirty settings are added on the fly to the shortcuts panel)

Lenbok, 2017-05-13:
The "original" behaviour is also what you would get (and what you still get) if you changed the layer height via a per-object override rather than per-plate override. So in that sense, the original behaviour is actually more consistent.

I would expect that that opening and closing profiles, and even editing profile properties that aren't included in the the current overrides (whether they be per-object or per-plate), should not change any of the override values.

The ambiguity is what to do if a profile is edited to change one of the properties that the user is employing as an object or plate override. I would think that this is actually a rare occurrence, and perhaps at that point it is worth asking the user if they want their current overrides updated.

foreachthing, 2017-05-13:
> It's - IMHO - not ergonomic because we are already used to test settings this way

Well, yeah, but new is not always bad! :-)
I like to have fast access to the settings I change and tweak until I'm happy.
So, _to me_, these override-settings should *not* be overridden and should be _master_ - no matter what's in the other settings.

dl6lr, 2017-05-18:
I vote for not changing overrides at all if you go into the profile editor with or without changes.
Another suggestion for the UI:
If you override parameters, i.e. layer height, display the effective (override) value in the profile dialog and mark them as changed. So the user can see the effective value in the dialog. This would additionally enable the user to save this setting as a new profile. This could be used for per-object overrides as well. And in my opinion this is intuitive to use, as the same behaviour is already used for changes in the profile witout saving them. 
One drawback is if you override several parameters but only want to change and save one of them to a new profile you have to reset them to the desired value before saving tem to a profile.

alexrj, 2017-05-20:
I appreciate all of your feedback, but I'd just like to point out that this behavior (re-syncing overrides with profile when closing the preset editor) was added after a number of people reported the opposite behavior as a confusing issue. :) I do see how both behaviors are potentially confusing, and maybe if I didn't change it the same people discussing this issue would be reporting the opposite one.

This is one more reason why I think the experiment of adding plater overrides partially failed, and the slight change towards a shortcuts pattern (#3912) is more streamlined.
By the way, we already have overrides for most settings - just double click on objects to add them.
More thoughts in #3912.

foreachthing, 2017-05-22:
@alexrj, then, don't call it "overrides" if they should/could/would not override settings.

Lenbok, 2017-05-22:
@foreachthing Pay attention and read #3912 then - that's exactly what that is about.

The current plater overrides are an awesome feature, giving the ability to easily add both one-off and commonly made adjustments on a per-print basis without dicking with your presets, and so represents a huge step forward from the existing per-object overrides in terms of usability and even functionality (you can now add overrides for settings that aren't available via per-object overrides). The current implementation of plater overrides is not without issues, but I don't think that they are anything particularly difficult.

However #3912 is set to neuter most of this step forward, by making the changes write back to your presets, transparently saving them, and removing the ability to add one-off overrides. It still represents an improvement over what we have in 1.2.9 but isn't as good as what it could be.

foreachthing, 2017-05-23:
@Lenbok; gotach!
To me, @alexrj, you can close here - thanks.
I'll jump to #3912 as well, if needed.

lordofhyphens, 2017-06-04:
Closing as @foreachthing has indicated what he wants is covered in #3912.

closed 2017-06-04 by lordofhyphens