slic3r/Slic3r #3946: Large chunk missing from layers compared to 3D view

larsrc, 2017-05-11:
### Version
1.3.0-dev Mac build from 2017-05-10

Config: 
[slic3r-config.ini.zip](https://github.com/alexrj/Slic3r/files/994345/slic3r-config.ini.zip)

### Operating system type + version
Mac OS 10.12.4

### Behavior
* _Describe the problem_
Model that looks nice in the 3D view has big sections missing in the sliced view.

* _Steps needed to reproduce the problem_
Load the Battery_dispenser_extension.stl from http://www.thingiverse.com/thing:399744/#files 
Slice it.

* _Expected Results_

The layers in the sliced view correspond to how it looks in the 3D view

* _Actual Results_
3D View: [image]

Sliced preview: [image]

#### STL/Config (.ZIP) where problem occurs

[Battery_dispenser_extension.stl.zip](https://github.com/alexrj/Slic3r/files/994334/Battery_dispenser_extension.stl.zip)

xoan, 2017-05-11:
The model has non-manifold errors:

[image]

lordofhyphens, 2017-05-11:
Confirmed on Windows that model is garbage. When you see "auto repaired", trust nothing. 
[image]

lordofhyphens, 2017-05-11:
http://service.netfabb.com

closed 2017-05-11 by lordofhyphens

larsrc, 2017-05-12:
So the "auto repaired" message should actually read "attempted repaired" - there's no guaranteed that it's repaired.