slic3r/Slic3r #3948: failing test for motionplanner.t

gege2b, 2017-05-12:
Hi all
I was unable to build slic3r today : 

### Version
_Version of Slic3r used goes here_
Current dev ( 1c4b8602cb7b79bc6fffbe12b7e5b71ac22900a7 )

### Operating system type + version

Linux Mint 18.1 and Kubuntu 16.04

### Behavior
* _Describe the problem_

Test fail for 18_motionplanner.t

```
(...)

[all is fine up to this, then I get these two warning that I think are not related,
but I put them here just in case]

src/admesh/stlinit.c: In function ‘void stl_read(stl_file*, int, int)’:
src/admesh/stlinit.c:296:36: warning: ignoring return value of ‘int fscanf(FILE*, const char*, ...)’, declared with attribute warn_unused_result [-Wunused-result]
       fscanf(stl->fp, "endsolid\n");
                                    ^
src/admesh/stlinit.c:297:40: warning: ignoring return value of ‘int fscanf(FILE*, const char*, ...)’, declared with attribute warn_unused_result [-Wunused-result]
       fscanf(stl->fp, "solid%*[^\n]\n");  // name might contain spaces so %*s doesn't work and it also can be empty (just "solid")
                                        ^
[... build continue and then]

t/18_motionplanner.t ............ 1/20 
#   Failed test 'path length is greater than straight line'
#   at t/18_motionplanner.t line 39.

#   Failed test 'path is fully contained in expolygon'
#   at t/18_motionplanner.t line 42.
# Looks like you failed 2 tests of 20.
t/18_motionplanner.t ............ Dubious, test returned 2 (wstat 512, 0x200)
Failed 2/20 subtests 
t/19_model.t .................... ok   
t/20_print.t .................... ok   
t/21_gcode.t .................... ok   
t/22_exception.t ................ ok   

Test Summary Report
-------------------
t/18_motionplanner.t          (Wstat: 512 Tests: 20 Failed: 2)
  Failed tests:  3, 6
  Non-zero exit status: 2
Files=21, Tests=507,  2 wallclock secs ( 0.09 usr  0.01 sys +  1.04 cusr  0.05 csys =  1.19 CPU)
Result: FAIL
Failed 1/21 test programs. 2/507 subtests failed.
FAIL
! Installing ./xs failed. See /home/gege/.cpanm/work/1494609722.4644/build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting
```

* _Steps needed to reproduce the problem_

git pull && perl Build.pl

gege2b, 2017-05-12:
OK, @alexrj was quickier than me and pushed 9670f2b712c509487235bd97dda859cf2c55ec97 while I was writing the ticket...

damn ! :)

closed 2017-05-12 by gege2b

lordofhyphens, 2017-05-12:
Usually you don't need to report test failures.