slic3r/Slic3r #3949: class MultiPoint hasn't virtual dtor

bmx666, 2017-05-12:
### Version
commit 1c4b8602cb7b79bc6fffbe12b7e5b71ac22900a7

### Operating system type + version
Ubuntu 16.04 x64

### GCC Warnings
[class MultiPoint hasn't virtual dtor](https://github.com/alexrj/Slic3r/blob/1c4b8602cb7b79bc6fffbe12b7e5b71ac22900a7/xs/src/libslic3r/MultiPoint.hpp#L20-L21)
```
warning: ‘class Slic3r::MultiPoint’ has virtual functions and accessible non-virtual destructor
```
fix **MultiPoint.hpp**
```cpp
virtual ~MultiPoint() {};
```

lordofhyphens, 2017-05-12:
1c4b860 had a test failure, it's probably using an older version of Slic3r's xs.so that is causing this exact error. Test failure has been fixed.

bmx666, 2017-05-12:
I knew about test failure. I just say you have issue in C++ class code in master branch!

lordofhyphens, 2017-05-14:
@bmx666 Which compiler are you using? GCC 6.3.0 with -Wall doesn't bring that warning up. 
[build.txt](https://github.com/alexrj/Slic3r/files/999761/build.txt)

bmx666, 2017-05-15:
@lordofhyphens I compiled with GCC 5.4.0 with flags:
- -Wall
- -Wextra
- -Weffc++
- -Wpedantic

Many useful warning flags in GCC disabled by default and not included in default warning flags like **-Wall** or **-Wextra**, many of them output false positive results or ouput many warnings from C-headers when compile for C++, but after analyzing help find hidden issues like that.

alexrj, 2017-05-15:
Thank you, I fixed this warning by defining a protected nonvirtual destructor. The MultiPoint base class is not supposed to be constructed and destructed directly.

closed 2017-05-15 by alexrj