slic3r/Slic3r #3950: Overhang handling suggestions

VanessaE, 2017-05-13:
Slic3r just does not handle overhangs very well, and fails to detect them reliably.  This needs fixed.

After doing a LOT of testing, I've made a list of overhang settings and behaviors that need to be added. I know you don't like adding new settings, but in this case I think it's unavoidable.  These should all be applied to (and only to) unsupported overhangs:

**[Print Settings]**
* overhang print speed - slower is better (to a point), gives the filament more time to cool.  Go slow enough and you can draw straight lines in mid-air (if you have enough fans).
* overhang acceleration - overhangs usually have soft or rounded corners because the nozzle simply skims around the corner too fast, causing the filament to pull inward from the fast movement and die swell.
* A checkbox to force concentric infill (or just a dropdown to select the pattern) for the bridging pattern and all bottom solid layers of the overhang.  Currently, one can force this by increasing the number of global perimeters to a point where rectilinear bridging is simply not generated, but that wastes filament.  It definitely produces better overhang surfaces, though.  My theory two-fold:  1) rectilinear patterns on such small areas concentrate too much heat into neighboring lines, whereas concentric spreads it out, and 2) the perimeters *will* droop some, providing nothing for the turn points of the central rectilinear pattern to anchor to in some cases and causing them to droop or warp sharply at the un-anchored edge, whereas concentric lines have long, parallel runs of neighbors to stick to, and in any case they will all droop by more or less the same amount.
* A checkbox indicating whether to print overhanging exterior perimeters first or last - counter-intuitively, exterior-perimeters-first produces much better overhangs for me.  If that means overriding the global first-or-last setting, criss-crossing the lines, and creating a seam somewhere, so be it.  Slic3r should of course use its same seam-placement logic here, so long as said seam isn't too far from the overhang.
* Threshold settings to decide how much of an overhang (as a percent or degrees, like the support overhang threshold) is needed before invoking these options - one for the flow ratio, and second one for the other options.

**[Filament Settings]**
* flow ratio - less may be better, for some filaments (think ABS versus PLA), and will likely need to be different from bridging flow ratio.
* fan speed during overhangs - some filaments can't have the fan at 100% all the time (again ABS versus PLA)
* Distance from the overhang at which the above fan speed and print speed begin/end - fans need time to spin-up.  It may be viable to use this to decide how far away the above criss-cross seam is allowed to get if Slic3r can't find a suitable, closer place first.

These two test models (especially the first one) are very good for evaluating these sorts of settings:
http://www.thingiverse.com/make:328122
http://www.thingiverse.com/thing:58218

This has been a thorn in the side of Slic3r users for quite some time now, in some cases for several years.  Related  issues:  #328, #2158, and #2313

(The gist of this issue comes from a now-redundant comment I made in #328)

VanessaE, 2017-06-05:
regarding the quality of the undersides of the overhangs, I guess a pic is warranted:

[image]

Not visible in this image is that the middle one (30mm/sec int and ext, 7 perims, min layer time of ~63 sedconds) also exhibits the least amount of upward curl/warping.

VanessaE, 2018-04-12:
Is someone going to address this?  It's been nearly a year.

supermerill, 2018-07-11:
Some questions:

Using "Detect bridging perimeter" should already decrease the speed & increase the flow.

* Overhang straight line speed : should be already done in "Detect bridging perimeter", or not?
* Overhang acceleration & Overhang corner speed : same. A specific "perimeter bridge acceleration" is really needed or the bridge acceleration is enough?
* Overhang line width : may be difficult to do. My idea was to do the opposite. On my list.
* Overhang exterior first threshold : it's like activating "external perimeter first" when i detect small overhang. This may create extrusion inconsistency for the external perimeter and be visible on the print (as "bad quality"). If done only for overhangs, these one need to be separated from the perimeter, a seem (or two) may be visible... i don't know if it's worth it.
* A checkbox to force concentric infill: I think that's basically what i done with my enhanced "Extra perimeters if needed" in #4476, but printed before the external perimeter instead of after. (if i understand well).
* Overhang flow ratio : should be possible, i'lll try.
* Overhang fan speed : difference with bridge fan speed?
* Minimum fan spin-up time : use a better firmware / modify it. Not the job of the slicer, imho. Also, it's impossible to know when a g-code command is done, as the printer can slow itself, or pause, or whatever.
* Height-over-bridge fan threshold : I'll see if it's possible. I've already added the opposite for infill ratio.

tl/dr:
easy things : Overhang flow ratio & Height-over-bridge fan threshold
hard thing: Overhang line width
For the opter ones, can you convince me?

foreachthing, 2018-07-11:
> Minimum fan spin-up time : use a better firmware / modify it. Not the job of the slicer, imho. Also, it's impossible to know when a g-code command is done, as the printer can slow itself, or pause, or whatever.

I wish this was a Slic3r "issue". My UM2 is _always_ late with higher fan speeds. When a small bridge (< 15 mm) is coming up, the fan has reached its speed about when the bridge has been passed. So, if I print bridges, I need the fan to be at 100% - from layer 0 to the end....

lordofhyphens, 2018-07-11:
@foreachthing #3308 Generally I agree though re: not being Slic3r's job to handle the minutae of fan speeds. 

There's a post-process script there though that should be helpful.

supermerill, 2018-07-11:
also kickstart  != spin-up time
kickstart is make it start at low rpm
spin-up time is to make it spin-up in advance.

VanessaE, 2018-07-11:
> Using "Detect bridging perimeter" should already decrease the speed & increase the flow.
> Overhang straight line speed : should be already done in "Detect bridging perimeter", or not?
> Overhang [...]

Overhangs are not bridges :smile: and really, nothing related to bridging should be used to handle overhangs, because they require an entirely different logic.

For example, a bridge segment (as defined in slicing terms) never turns a corner or goes around a curve; the perimeters are always long, straight line segments.  Since it's unsupported, it's always drawn at "full" flow (or close to it, per the configured bridging flow ratio).  Bridges necessarily always use rectilinear infill, though not always oriented ideally.  Since they're always straight, a bridge can be drawn quite fast compared to an overhang, and one can exploit the filament's die swell and shrinkage characteristics in a bridge, as well, where one definitely wants to minimize their effects when drawing an overhang.

Overhangs that are especially steep (about 65° from vertical) do currently use what I assume is bridging flow, but shallower overhangs are just drawn as a simple wall.  I now don't recall if bridging speed is ever used on an overhang.  Overhangs don't seem to get any extra fan at all, regardless of angle.

> Overhang line width : may be difficult to do. My idea was to do the opposite. On my list.

For larger parts, wider perimeter lines definitely produce better-quality overhangs, as seen in my photos.  For small, detailed overhangs on super-thin layers (such as the edges of the front window in https://www.thingiverse.com/thing:2719434), narrower lines seem to be better.

> Overhang exterior first threshold : it's like activating "external perimeter first" when i detect small overhang. This may create extrusion inconsistency for the external perimeter and be visible on the print (as "bad quality"). If done only for overhangs, these one need to be separated from the perimeter, a seam (or two) may be visible... i don't know if it's worth it.

Although I had it in mind to do this per-island, per-layer (so not switching order mid-perimeter), it occurs to me that it may be better to do this on a whole-mesh basis, so that the whole mesh has its perimeters first or not, for consistency.

That way, if you have a dozen assorted meshes on the plate (whether as individual files, or as copies, if a single .STL brings in several meshes), but only one mesh needs this special treatment, then it would get handled automatically.  Basically, always assume that the user can't change the existing global setting (in the per-part "Settings" dialog) without ruining the project.

> A checkbox to force concentric infill: I think that's basically what i done with my enhanced "Extra perimeters if needed" in #4476, but printed before the external perimeter instead of after. (if i understand well).

That looks like it covers it, yeah.

> Overhang fan speed : difference with bridge fan speed?

As above.  A bridge generally requires 100% fan speed (if the filament allows fan at all), while an overhang might do best with say 50%, with the rest of the part printed with say 10% fan.

>Minimum fan spin-up time : use a better firmware / modify it. Not the job of the slicer, imho.

As mentioned above kick-start is not spin-up time.  It's just a brief 100% burst to get over the "hump" to get a fan to start spinning at a low final speed.

Has to be done in the slicer - while it's possible to program a firmware to know how long a fan takes to spin up, and the firmware can look ahead in its receive buffer to predict and adapt to some things (such as changes in flow rate, as in Linear Advance), it's impossible for the printer to know when to start spinning up the fan in a real print.

On a complex or detailed model, you might have a couple dozen line segments between where the fan needs to start spinning up, and the bridge/overhang where the extra airflow is needed, but the firmware only buffers maybe 8 line segments, so even if it did look ahead, by the time it finds the fan command and its successive bridge or overhang segment, it's already too late to start spinning it up.

> For the other ones, can you convince me?

I hope I did :smile:

supermerill, 2018-07-11:
>nothing related to bridging should be used to handle overhangs, because they require an entirely different logic.

They both draw "unsupported" feature. I will test with reduced speed & acceleration when printing your first test object to convince myself.

> For larger parts, wider perimeter lines definitely produce better-quality overhangs

Do you know why? (also , as it's unsupported, this extrusion should be round, so wider & deeper) DO you used "Detect bridging perimeter" when making these tests?

> That way, if you have a dozen assorted meshes on the plate (whether as individual files, or as copies, if a single .STL brings in several meshes), but only one mesh needs this special treatment, [...]

You can use a mesh setting for that. I don't see the need to do it automatically (it can surprise the user in a bad way).

> A bridge generally requires 100% fan speed (if the filament allows fan at all), while an overhang might do best with say 50%

Really? Do you have some stl/filament combination to back that?

> It's impossible for the printer to know when to start spinning up the fan in a real print.

It is, that's the purpose of the cgode. I say to the printer to have the fan at X% at this moment of the print. I will try to make something on klipper.

lordofhyphens, 2018-07-11:
@VanessaE @supermerill 

Just stepping in to ask people to confirm terms (to make sure people aren't talking past each other). Thanks.

VanessaE, 2018-07-11:
> [...] confirm terms [...]

For the sake of this thread:

"Overhang" means any section of the print that temporarily leads out at an angle from a vertical wall, partly over air, turns one or more corners while out there, and has start and end points in the same general region of the print.

A bridge is a completely unsupported section that just stretches straight across the void between two opposing anchor points.

It is of course possible for an overhang to lead to a bridge, or to stretch a bridge between two overhangs, as in the case of the uppermost few layers that complete a horizontal hole.

> > For larger parts, wider perimeter lines definitely produce better-quality overhangs
>
> Do you know why? (also , as it's unsupported, this extrusion should be round, so wider & deeper) Do you used "Detect bridging perimeter" when making these tests?

I am not sure why it works, but it does -- I'm not a fluid dynamics expert. :smile:

Since most overhangs are less than the aforementioned 60° threshold (or 65°, or whatever it actually is), such overhanging extrusions are not round - they're partly flat, because at least part of the width of each line is supported by and squished down onto the layer underneath.

And yes, I always have "detect bridging perimeters" enabled.

> >That way, if you have a dozen assorted meshes on the plate [...]
>
> You can use a mesh setting for that. I don't see the need to do it automatically (it can surprise the user in a bad way).

I suppose so, but asking the user to create/use a modifier mesh is a LOT more complicated than just having them turn on/off a checkbox.  No, there's too much focus on modifier meshes, when the program could just do some stuff automatically.

To be honest, although they are useful, they're turning into a crutch, just as post-processing scripts are.

> > A bridge generally requires 100% fan speed (if the filament allows fan at all), while an overhang might do best with say 50%
> 
> Really? Do you have some stl/filament combination to back that?

It was a hypothetical example, however:

As I write this, I'm printing a new revision of the MK8 gear cartridge for my extruder, using Atomic Bright White PETG.  This combination does well with a continuous 25% fan.  As you may know, PETG doesn't bridge worth a damn without 100% fan, and it likes a goodly amount of airflow on overhangs as well, but not necessarily 100%.  It's possible to "over-cool" PETG, leading to curling and warping, enough to pop it right off the plate.

> > It's impossible for the printer to know when to start spinning up the fan in a real print.
>
> It is, that's the purpose of the [g-code]. I say to the printer to have the fan at X% at this moment of the print. I will try to make something on klipper.

The point is, if you insert `M106 S255` just before a bridge line segment, of course you're demanding "100% fan *now*... but your if fan is slow to spool up, then it won't reach anywhere near full blast until somewhere partly through the bridge, if it gets there at all.  My fans, for example, take a bit over 1 second to get from 25% to 100%, and over 2 seconds to go from 0% to 100%, so smaller bridges basically get no extra fan, even though they need it.

*All* bots using commonly-available components have this problem.  The only way to combat that is to turn the fan up early.  That is, unless you're using a valve-controlled hose leading to a big, remotely-mounted, always-on fan, or a tank and air compressor. :stuck_out_tongue: 

> > I hope I did
> 
> Not yet ;-p

I guess I need to try harder :stuck_out_tongue:

supermerill, 2018-07-11:
I've made a test:
[image]

first is default settings + Detect bridging perimeter
second is the same than 1 + my "Extra perimeters if needed"
third is the same than 1 + i reduced the speed/acceleration from 20/1000 to 5/100

for the fan speed start-up time, I've addressed that here : https://github.com/KevinOConnor/klipper/issues/474

VanessaE, 2018-07-11:
Well that's at least consistent with my initial experiment.

curiouspl2, 2018-07-26:
Hello. 
Got new idea about handling overhang volume compensation (because it is one of main issues causing overhangs to be printed with not enough material , causing them to curl up)

Main problem of handling overhangs currently is that it is difficult to tell if overhang is 'unsupported' - geometries of vector paths on both layers need to be taken under account. Analysis of gcode is difficult and does not allow to f.e. change arc or circle to a segmented object easily, 
so all the flow/volume correction should be done before gcode export phase.

Also extrusion widths vary. It might happen that extrusions below path we analyze have weird shapes (like being perpendicular to our path) and all this makes computation and segmenting complex, and devels pulling out their hair out, while attempting to code algorithms for inter-layer analysis...

but there is simpler path :)

after slicing there is volumetric path (not yet gcode) being created for each layer - used also for preview. 
It contains data about tool path and volume of extrusions. 
From this data 2d array of volume can be computed, for each layer. Note we do not need to keep 
all such layers, we just need to keep two of them, for adjacent layers, in memory. 

Having such crude volume maps for two layers , 'correction' layer can be computed using various methods. Just difference and threshold tells much. Areas with no support will mean bridging and are 
solved by bridging algo. Rest will compute 'correction' layer map .

Now this map can be used to segment and correct overhang extrusions at next layer, as it represents
missing volume in Z plane. This means this volume can be added to toolpath , and this added volume
does not change width of the extruded lines (because volume will fill voids layer below).
Note it does not involve comparing geometries, and even allows situations like partial corrections, like 
in case when single map pixel will span over f.e. two perimeters, correction of first perimeter will subtract just part of correction from the pixel and correction for another perimeter will subtract another.

Also as we are not dealing with gcode, we know width and extrusion volume of each feature, so 
we can simply apply corrections and repeat process with next two layers. 
Also geometries of artifacts causing volume correction do not matter, just fact of missing volume in 
specified area. 

And as another benefit - array map can be of any defined resolution, so it is predictable computationally (does not depend on model size) and allows corrections to be computed in reasonable time. If resolution will be changed, this time will change aswell, allowing to choosing memory/time vs correction accuracy compromise by user.

I would like to hear more comments on that as this is one of very pressing issues causing
even simple prints to fail and having only very tedious workaround thru modifier meshes route which 
also breaks perimeters, so produces very ugly results.

supermerill, 2018-07-27:
@curiouspl2

> change arc or circle to a segmented object easily,

stl & most 3D models store only triangles, so no "arc" or "circle" can exist.

Do you propose something similar to my "over-bridge compensation" in my fork, but also for overhangs?

toskium, 2019-02-06:
Anything new on this issue yet? I see this being requested and marked duplicate to the original issue but nothing regarding putting this on the development roadmap.

supermerill, 2019-02-06:
@toskium see https://github.com/slic3r/Slic3r/pull/4717

zbrozek, 2019-02-17:
I'm struggling to print some parts in ABS that have overhangs right now. In particular I would *love* to have overhang-dependent fan speed. My parts are super brittle if I print with fan all the time, but there's no way to get a decent overhang without it.

Much like we have filament cooling options based on layer time, I'd love to have them based on "supportedness" too.

ciordia9, 2020-07-23:
Still relevant as the years tick by.

VanessaE, 2020-07-23:
@supermerill has implemented a number of these changes in his [SuperSlicer](https://github.com/supermerill/SuperSlicer) fork.

ciordia9, 2020-07-23:
@VanessaE Not to bug-jack, I've been looking for specifically a fan power during overhangs for ABS. I've got a current build of SS but I didn't see it in there. Have to read through SS a bit more carefully to see what else of your requests have been baked in. Good stuff!

pass75, 2022-01-12:
Vanessa, I landed to this this post because I was looking for the same specific overhangs settings in PrusaSlicers. I totally agree with you that overhangs are probably the worst scourge of 3d printing and are underestimated by slicing programs. I recently discover some of them were implemented in SuperSlicer. Thank you for your efforts in supporting this. One note: reading your posts I was impressed by your skills and competences: you should definitely take part in SuperSlicer developer team! Greetings from Italy.