slic3r/Slic3r #3951: Bridge perimeters printed with perimeter speed, #657 was closed but problem remains

perkmeister, 2017-05-13:
### Version
1.34.1-prusa3d-win64  

### Operating system type + version
Windows 7 64 bit

### Behavior
Perimeters are not detected as bridges and are printed at perimeter speeds.  This is the same behaviour as reported in #657 which was closed as being fixed, but unfortunately is still alive. It is particularly bad for PETG as these need to be printed at low speeds with bridges 

#### STL/Config (.ZIP) where problem occurs
If you try to print the attached the perimeters around the bridges in the middle of the object get printed first at high speed, whereas the bridges section will be at the correct bridging speed. This will result in a messy goo on your print with PETG. It appears Slic3r is not recognising these perimeters as needing to be printed at bridge speeds, although it does appear some outer perimeters are detected correctly.

[perimeter_not_bridging.zip](https://github.com/alexrj/Slic3r/files/998811/perimeter_not_bridging.zip)

lordofhyphens, 2017-05-13:
You need to provide a reproduction case with this version of Slic3r, not the Prusa3d version.

I will reopen if you do so.

perkmeister, 2017-05-13:
OK, I assumed they were based on the same underlying code. I have Slic3r installed too, and the problem exists with this as well. This is with Slic3r 1.3.0-dev. See attached zip.
[Perimeter_bridge_slic3r.zip](https://github.com/alexrj/Slic3r/files/998831/Perimeter_bridge_slic3r.zip)

lordofhyphens, 2017-05-13:
@perkmeister They have bugs/features we don't and vice-versa. I've updated the issue template to make this clear.

alexrj, 2017-05-16:
@perkmeister, thank you for this report. The issue was actually fixed, but small_perimeter_speed was overriding the overhang/bridge speed since some perimeters of your test model are small. I now changed this behavior and added regression tests for overhang perimeters.

closed 2017-05-16 by alexrj

perkmeister, 2017-05-16:
Excellent, thanks!