slic3r/Slic3r #3955: Concentric infill error with narrow members in lattice structure

lilatomic, 2017-05-15:
### Version
1.3.0-dev, hash b826f8b

### Operating system type + version
Windows 10 Home 64 bit, version 1703 (build 15063.296) 

### Behavior
Concentric infill is not produced when used on lattice structures (with narrow members).
Despite concentric infill being selected and infill density set to 100%, a rectilinear raster path is generated (figure 1).
Figure 1
[image]

_Steps to reproduce_
* Set infill density to 100%, set infill pattern to concentric
* Set number of perimeters to 1

_Expected Results_
It is expected that the fill will be concentric (figure 2), which in this case maximises part strength and build quality. 
Figure 2
[image]

 _Actual Result_
The fill is not concentric, but is a rectilinear raster pattern (figure 1). This causes unnecessary back-and-forth of the extruder head on the narrow members, which leads to less strength and poor print quality.

_Is this a new feature request?_
This is a very specific bug report, so it's probably more of a feature.

#### STL/Config (.ZIP) where problem occurs
[lattice_infill_issue.zip](https://github.com/alexrj/Slic3r/files/1000169/lattice_infill_issue.zip)

#### Work
I am aware that increasing the number of perimeters can help alleviate this issue, but this can interfere with parts of the model which are not lattice members. Modifier meshes can be used to localise the effects to the lattice areas, but it does not make the vertical struts concentric.
I've been hired as a research assistant for the Additive Design & Manufacturing Laboratory at McGill University, and one of my tasks for the summer is to fix this. From my understanding of the code, this is caused not by the FillConcentric. I think it's being cause by the gap filling, which I guess isn't pulling the right infill style? I still need to trace this, and any help would be appreciated :)

lordofhyphens, 2017-05-15:
* Solid infill that isn't top infill is always rectilinear, as rectilinear is generally the only pattern that is reliable at 100% (which is often a waste of material).
* Gap fill is not filled with a pattern.
* Fig2 shows extra perimeters (red) and rectilinear fill (blue)
* Fig1 looks like fewer perimeters.

lilatomic, 2017-05-15:
Even at different levels of infill density, this still occurs. Here's it at 40% infill density:
Figure 3
[image]
The only thing that changes with different infill density levels is that some layers become more like the desired pattern in the struts: 
Figure 4
[image]
Note that the vertical struts are still using a rectilinear fill

With regards to the 3rd and 4th points, I agree that is what it looks like, but the settings are configured for concentric infill.
Another interesting thing to note is that I can find figure 3 appearing for all infill densities (using the same settings as before with different infill densities) at layer height 21.35.

lordofhyphens, 2017-05-15:
All of those infill areas are rather small, you should reduce the solid infill threshold value.

lilatomic, 2017-05-17:
Thanks for the suggestion. But changing the solid infill threshold value doesn't seem to change anything, so I'm not sure what's up with that. Here's with infill threshold of 0.001, 1, and the default 70:
Figure 5:
[image]
[image]
[image]

I modified the lines where it defaults to ipRectilinear to instead default to ipConcentric and built it (on a Ubuntu machine, I was having trouble building wxWidgets on Windows), and it seems to work well in some circumstance (figure 6), but not in others (figure 7):
Figure 6
[image]
Figure 7
[image]

It also fails on build test t\\fill.t 94/95 'no missing parts in solid shell when fill_density is 0', which I guess makes sense given how the struts in Figure 7 look. 
Why Concentric generates those weirdly segmented struts, but increasing the perimeter doesn't, I'm not sure. It seems they both use offset2 (if thin_walls is configured to true), but with slightly different parameters. I will look into it further, but I'd appreciate any knowledge you'd be willing to share.