slic3r/Slic3r #3957: German special characters (ö, ä, ü) in STL filename can not be opened in Slic3r

Quas7, 2017-05-16:
### Version

tested with:
slic3r-master.2017.05.14.1042.b826f8b.64bit
slic3r-master.2017.04.27.978.9bdc537

1.3.0-dev

### Operating system type + version
Windows 10

### Behavior
GUI can not load STL-files with filenames containing German special characters (ü, ä, ö).

"Error while opening t''wers.stl: Error while reading STL file" 
(äüö get displayed as different ASCII characters in the error message )

correct filename is "töwers.stl"

[image]

[towers.zip](https://github.com/alexrj/Slic3r/files/1004090/towers.zip)

lordofhyphens, 2017-05-16:
#2955 
#3923

closed 2017-05-16 by lordofhyphens

lordofhyphens, 2017-05-18:
@Quas7 Recommend you follow https://github.com/alexrj/Slic3r/issues/2955 to get updates.

alexrj, 2017-05-19:
This was fixed now. It can be tested with the latest dev build available at http://dl.slic3r.org/dev/win