slic3r/Slic3r #3959: Notes in profiles not showing in G-code headers.

bah2830, 2017-05-17:
### Version
1.3.0

### Operating system type + version
Mac OSX 10.12.4

### Behavior
* Since the addition of Notes to profiles it has a tooltip 
**"You can put here your personal notes. This text will be added to the G-code header comments."**
  * This does not happen. I am unable to find my notes in the g code.

I have found a human error with slicing multiple objects coming up and don't have a way to see what settings and filament profile was used so I forget which filament I need for that object over time.

I have worked around this by modifying the output filename but putting many details in it makes a very long filename.

If the notes are not meant to be in the G-code the interface should probably be updated to reflect that in the tooltip, currently it says it WILL be in the g code.

lordofhyphens, 2017-05-17:
Currently the interface is putting notes at the bottom with the rest of the configuration options, at least.

bah2830, 2017-05-17:
Also this may require another issue.

When using [filament_preset] or [print_preset] in my custom g code I get an unexpected output. The same thing is happening in the file output name.

Custom start g-code
```
; Filament - [filament_preset]
; Print - [print_preset]
```

Output in G-code
```
; Filament - ARRAY(0x7ff6c6170948)
; Print - ARRAY(0x7ff6a813cb40)
```

lordofhyphens, 2017-05-17:
@bah2830 yes put that in another issue.

lordofhyphens, 2017-06-04:
Implemented via 25202f9a474eae9874ef3c0431bd0cdbba3bc963 and e064fdf17017fc4a51e7e029dc2a8464a4e6d6f2. Enjoy.

closed 2017-06-04 by lordofhyphens

lordofhyphens, 2017-06-04:
[image]

lordofhyphens, 2017-06-04:
So if you have multiple extruders will print those notes too (in extruder order), as well as the printer notes.