slic3r/Slic3r #3961: Feature Request: Password lock slicing profiles

curieos, 2017-05-17:
With shared printers, safe g-code is important. It is easy for some eager newcomer to get enthusiastic about "improving" the generated g-code of a printer and without supervision modify something, causing damage to the printer or overwriting the previous profile.

I think it would be a good feature to have an option to password lock the slicing profiles for these scenarios. There could be a file menu option to setup a password (and maybe a new option in the startup wizard?), and then the profiles could be locked with a toggle button (next to the save and delete buttons). If a user wants to modify them they can click the lock/unlock button and it would prompt them for a password to change the settings.

A step further (not sure if its necessary, this is up for discussion), maybe the .ini files could be password locked too? Looking into it on the linux side, I found some programs that can encrypt files with a password like gnupg.

lordofhyphens, 2017-05-17:
I don't think this is something that is in scope for Slic3r; it is almost always better to use a better solution (SVN with needs-lock comes to mind) than to try to implement a minimal security system. 

I keep my own Slic3r configuration under version control.

curieos, 2017-05-17:
An example, the makerspace I use to be apart of is having a problem with this. I'm not sure if they are using Slic3r or KISSlicer, but they have the tuned profiles backed up and every time the computer is turned on or the slicer is quit these profiles are restored. This does not prevent a user from tweaking the profiles while the slicer is running AFAIK, since this data is kept in memory and the config file is only read when switching between profiles. This is the problem that I see and I dont know any other way to fix it.

bmx666, 2017-05-18:
It's remind me some feature from customer:
1. At the beginning: I wanna lock settings by some key or password
2. After few days: I wanna lock all settings, but some of them available for modify without request access
3. After few weeks: I wanna manage control. Lock some parts of settings by roles, groups and audit every options and configure them with manage control panel.
4. After month: I wanna delegate some specific options for change by roles or groups.
5. Project was completed. How users using this feature now? Company create one master role and group, then delegate everybody all rights.

P.S. Useful feature for very big corps.

xoan, 2017-05-18:
@bmx666 lol :D

I agree this kind of features should be handled by the system itself or by using external tools, such rsync/vcs and permissions.

curieos, 2017-05-18:
How does this solution prevent the settings stored in memory from being modified? This is the real problem I'm chasing here. Those external tools may work fine to keep users from modifying the .ini files, but how does that stop them from opening slic3r and messing around?

bmx666, 2017-05-18:
@curieos, what about fork this project and hide buttons in gui, for prevent of modify settings?
But in console version users can use any settings.

curieos, 2017-05-18:
I understand the console can be used to do anything, thats fine. I only care about the UI.

This is a really simple problem with a really simple solution and tbh I think you are over complicating it. If I get time I will try to work on this on my fork.

7eggert, 2017-05-19:
What/how is your printing process and how do you intend the password from preventing damage while also allowing the necessary adjustments to the printer settings (brim, support, fan)?

alexrj, 2017-05-19:
@curieos, the scenario you described makes sense a lot. I have a good idea for solving it (no passwords involved but should still work). As soon as I can, I'll implement a proof of concept.

NateTG, 2017-05-19:
What about having some kind of 'profile history' so that it's easy to undo changes instead?

curieos, 2017-05-19:
I think some people were misinterpreting the problem so I rewrote my feature request to describe the scenario/problem that I see.