slic3r/Slic3r #3962: Add option to gradually cool extruder and bed temperature over X # of layers for more precise control?

JoelLisenby, 2017-05-17:
**Feature Request**

**Filament Settings**

Can we add an option to gradually cool extruder and bed over X # of layers for more precise temperature control?

Example Settings:
```
**Temperature (C)**

Extruder:
Starting Temp: [   ]
Ending Temp:  [   ]
Cooling duration (number of layers): [   ] (default 1)

Bed:
Starting Temp: [   ]
Ending Temp:  [   ]
Cooling duration (number of layers): [   ] (default 1)
```

lordofhyphens, 2017-05-17:
So what problem are you trying to solve with this feature request? We prefer to start from a problem statement, rather than jumping right to a solution.

JoelLisenby, 2017-05-17:
Thanks, I've updated my original post with the reason.

alexrj, 2017-05-21:
I'm not very convinced. It's good to have different temperatures between 1st layer and 2nd layer, for better adhesion while printing 1st layer. But why would you have different temperatures between 2nd layer and, say, 10th layer?

Anyway, this will be possible with conditional G-code and arithmetic parser (#3390).