slic3r/Slic3r #3965: Random segfaults and memory errors

flannelhead, 2017-05-18:
### Version
A fresh build from master, 5f6214b645d3d688d50fb157fccb409c1cc6661c

### Operating system type + version
Arch Linux x86_64, Linux 4.9.28

### Behavior
* _Describe the problem_
Segfaults or other random memory errors occur with the attached STL and settings.

* _Steps needed to reproduce the problem_
1. Open the attached STL with the attached settings.
2. Switch to the Layers tab from 3D to trigger slicing.
3. Open the print settings and change the number of perimeters to 4. This isn't the only way to trigger the issue; other changes to settings that trigger reslice also work.
4. Slic3r now automatically switches back to 3D tab.
5. Switch to Layers tab. Slicing is triggered and a segmentation fault happens.

I could reproduce this with other STLs too, but just attaching this one which I know for sure to reproduce the issue.

* _Expected Results_
Slic3r doesn't crash.

* _Actual Results_
Slic3r crashes.

#### STL/Config (.ZIP) where problem occurs
[slic3r_debug.zip](https://github.com/alexrj/Slic3r/files/1010920/slic3r_debug.zip)
Config, STL and possible useful `gdb` output is attached.

flannelhead, 2017-05-18:
I also managed to reproduce the crash on Windows 10 with [this build](http://dl.slic3r.org/dev/win/slic3r-master.2017.05.17.1053.5f6214b.64bit.zip) and a fresh config generated by the wizard using default values. I performed exactly the same steps.

flannelhead, 2017-05-19:
I had a `git bisect` session and found 9b4fe076a652864d54cd277ca06f40ce5cf94cec to be the first commit where this issue is present. Now I don't know if this commit is the root cause of the issue or does it just cause a previously introduced memory management error to manifest itself.

flannelhead, 2017-05-19:
A interesting note: the problem doesn't appear when running under `valgrind`. I wonder what this might imply...

/cc @alexrj

flannelhead, 2017-05-19:
Another note, this doesn't seem to appear if I have the "Background processing" option enabled in the Preferences.

gege2b, 2017-05-20:
I cannot reproduce this with current head on mint serena using the files you provided

flannelhead, 2017-05-20:
@gege2b This seems to be a memory management error, so it may manifest itself in different ways on different systems. I can almost always make Slic3r crash by repeating the mentioned sequence of actions, but the cause of the crash is different every time, but it is always some kind of a memory-related error.

Since I'm able to repro this on my computer, I'm also offering my help to the devs - please tell which kind of information would you need.

Lenbok, 2017-05-23:
While I haven't tried with your config files, I also get very similar segfaults while doing actions quite similar to those you describe -- essentially it seems the crashes are triggered when switching back to the layers tab while a reslice is in progress.

Example GDB trace from one of mine (I am on ce68851):
````
[...]
[Thread 0x7fffd2b41700 (LWP 7043) exited]
Background processing completed.
No background process running.
[New Thread 0x7fffd2b41700 (LWP 7146)]
Background processing started.
Starting thread 3 (parent: 0)...
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

Thread 1 "perl" received signal SIGABRT, Aborted.
0x00007ffff7319428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) info threads
  Id   Target Id         Frame
* 1    Thread 0x7ffff7fba700 (LWP 6888) "perl" 0x00007ffff38231f1 in XS_Wx__App_MainLoop (my_perl=0x7d2010, cv=<optimized out>) at Wx.c:15086
  2    Thread 0x7fffe9a69700 (LWP 6904) "dconf worker" 0x00007ffff73deb5d in poll () at ../sysdeps/unix/syscall-template.S:84
  3    Thread 0x7fffe9268700 (LWP 6906) "gmain" 0x00007ffff73deb5d in poll () at ../sysdeps/unix/syscall-template.S:84
  4    Thread 0x7fffe8a67700 (LWP 6907) "gdbus" 0x00007ffff73deb5d in poll () at ../sysdeps/unix/syscall-template.S:84
  55   Thread 0x7fffd2b41700 (LWP 7146) "perl" __GI___libc_malloc (bytes=48) at malloc.c:2909
(gdb) thread apply all bt

Thread 55 (Thread 0x7fffd2b41700 (LWP 7146)):
#0  __GI___libc_malloc (bytes=48) at malloc.c:2909
#1  0x00007ffff5460e78 in operator new(unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x00007ffff59d6efa in __gnu_cxx::new_allocator<std::_Rb_tree_node<std::pair<std::pair<int, int> const, int> > >::allocate (this=<optimized out>, __n=<optimized out>)
    at /usr/include/c++/5/ext/new_allocator.h:104
#3  std::allocator_traits<std::allocator<std::_Rb_tree_node<std::pair<std::pair<int, int> const, int> > > >::allocate (__a=..., __n=<optimized out>) at /usr/include/c++/5/bits/alloc_traits.h:491
#4  std::_Rb_tree<std::pair<int, int>, std::pair<std::pair<int, int> const, int>, std::_Select1st<std::pair<std::pair<int, int> const, int> >, std::less<std::pair<int, int> >, std::allocator<std::pair<std::pair<int, int> const, int> > >::_M_get_node (this=<optimized out>) at /usr/include/c++/5/bits/stl_tree.h:491
#5  std::_Rb_tree<std::pair<int, int>, std::pair<std::pair<int, int> const, int>, std::_Select1st<std::pair<std::pair<int, int> const, int> >, std::less<std::pair<int, int> >, std::allocator<std::pair<std::pair<int, int> const, int> > >::_M_create_node<std::piecewise_construct_t const&, std::tuple<std::pair<int, int> const&>, std::tuple<> >(std::piecewise_construct_t const&, std::tuple<std::pair<int, int> const&>&&, std::tuple<>&&) (this=<optimized out>) at /usr/include/c++/5/bits/stl_tree.h:545
#6  std::_Rb_tree<std::pair<int, int>, std::pair<std::pair<int, int> const, int>, std::_Select1st<std::pair<std::pair<int, int> const, int> >, std::less<std::pair<int, int> >, std::allocator<std::pair<std::pair<int, int> const, int> > >::_M_emplace_hint_unique<std::piecewise_construct_t const&, std::tuple<std::pair<int, int> const&>, std::tuple<> >(std::_Rb_tree_const_iterator<std::pair<std::pair<int, int> const, int> >, std::piecewise_construct_t const&, std::tuple<std::pair<int, int> const&>&&, std::tuple<>&&) (__pos=..., this=0x7fffd2b40550) at /usr/include/c++/5/bits/stl_tree.h:2170
#7  std::map<std::pair<int, int>, int, std::less<std::pair<int, int> >, std::allocator<std::pair<std::pair<int, int> const, int> > >::operator[] (this=this@entry=0x7fffd2b40550, __k=...)
    at /usr/include/c++/5/bits/stl_map.h:483
#8  0x00007ffff59d7d66 in Slic3r::TriangleMeshSlicer<(Slic3r::Axis)2>::TriangleMeshSlicer (this=0x7fffd2b405e0, _mesh=<optimized out>) at src/libslic3r/TriangleMesh.cpp:1426
#9  0x00007ffff5a17949 in Slic3r::PrintObject::_slice_region (this=this@entry=0x3faa110, region_id=region_id@entry=0, z=std::vector of length 90, capacity 90 = {...}, modifier=modifier@entry=false)
    at src/libslic3r/PrintObject.cpp:810
#10 0x00007ffff5a1b30a in Slic3r::PrintObject::_slice (this=0x3faa110) at src/libslic3r/PrintObject.cpp:690
#11 0x00007ffff58de698 in XS_Slic3r__Print__Object (my_perl=<optimized out>, cv=<optimized out>)
    at /usr/bin/perl -MExtUtils::XSpp::Cmd -e xspp -- -t "/home2/len/dev/Slic3r.git/xs/xsp/typemap.xspt"  "/home2/len/dev/Slic3r.git/xs/xsp/Print.xsp":524
#12 0x00000000004bd3ca in Perl_pp_entersub ()
#13 0x00000000004b6276 in Perl_runops_standard ()
#14 0x000000000043c4ee in Perl_call_sv ()
#15 0x00007ffff61997de in S_jmpenv_run (my_perl=my_perl@entry=0x66508e0, action=action@entry=0, thread=thread@entry=0x3fc6990, len_p=len_p@entry=0x7fffd2b40f14,
    exit_app_p=exit_app_p@entry=0x7fffd2b40f0c, exit_code_p=exit_code_p@entry=0x7fffd2b40f10) at threads.xs:497
#16 0x00007ffff61999ef in S_ithread_run (arg=0x3fc6990) at threads.xs:591
#17 0x00007ffff76b46ba in start_thread (arg=0x7fffd2b41700) at pthread_create.c:333
#18 0x00007ffff73ea82d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 4 (Thread 0x7fffe8a67700 (LWP 6907)):
#0  0x00007ffff73deb5d in poll () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007ffff13f438c in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff13f4712 in g_main_loop_run () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
[... cut ...]
#23 0x00007ffff271e353 in wxEventLoopBase::Run() () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#24 0x00007ffff26e3666 in wxAppConsoleBase::MainLoop() () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#25 0x00007ffff38694e9 in wxPliApp::MainLoop (this=0x174d920) at cpp/app.h:195
#26 0x00007ffff38231f1 in XS_Wx__App_MainLoop (my_perl=0x7d2010, cv=<optimized out>) at Wx.c:15086
#27 0x00000000004bd3ca in Perl_pp_entersub ()
#28 0x00000000004b6276 in Perl_runops_standard ()
#29 0x0000000000443c59 in perl_run ()
#30 0x000000000041cbbb in main ()
(gdb) 

````

lordofhyphens, 2017-06-02:
If it's switching back to layers tab while a slice is in progress, I would think that Slic3r is trying to execute a draw command while the slicing is underway (bad data).

gege2b, 2017-06-02:
I get the same error

On my first test a few day ago, I made a little mistake : I switched to the PREVIEW tab instead of the LAYER one...

I've re tested right now (with current head) and indeed, hard crash with this error : 

```
Range iterator outside integer range at /home/gege/Documents/build/Slic3r/lib/Slic3r/Layer.pm line 31.
Perl exited with active threads:
        1 running and unjoined
        0 finished and unjoined
        0 running and detached
```