slic3r/Slic3r #3966: Suggested Option: Minimum Speed

ProfDrYoMan, 2017-05-19:
Using slicer-dev.

Is this a new feature request?
Yes.

In speed print settings %-values may lead to very slow speeds.
I suggest to add a minimum print speed.

E.g.:

Perimeters is 60 mm/s
Small perimeters is 20 mm/s
External perimeters is 30 mm/s

First layer speed is 50%

Now on the first layer small perimeters are printed with 10 mm/s and external perimeters with 15 mm/s. Such low speeds give me worse results than meaningful faster ones.

A minimum speed would limit the low speeds.

Minimum speed set to 20 mm /s and both perimeters would be printed with 20 mm/s.

cerdeira, 2017-05-19:
This request seems far too drastic.
It will cut other people choices when you can perfectly choose whatever you want today.

ProfDrYoMan, 2017-05-19:
Well, then I can choose from two options:

Setting first layer speed to 20 mm/s and everything is printed with slow 20 mm/s.
Setting first layer speed to 50% and a lot of stuff is printed a lot slower than needed.

Seems not like a win-win.

I guess minimizing the print times is what everybody wants.

alexrj, 2017-05-19:
What's wrong with setting First Layer Speed to 30mm/s?

ProfDrYoMan, 2017-05-19:
Then the small perimeters got printed with 30 mm /s but I wanted (for quality) 20 mm/s.

My values where just an example.

cerdeira, 2017-05-19:
Ok.
My point was, can you get what you want with current options?
If you can, no need to changes.

ProfDrYoMan, 2017-05-19:
No, I cannot get what I want with the current options.
I have to decide between two not optimal choices: https://github.com/alexrj/Slic3r/issues/3966#issuecomment-302744032

cerdeira, 2017-05-19:
So may be the solution would be to have a complete set of options for 1st layer speed.

ProfDrYoMan, 2017-05-19:
Yes. But this is clearly overkill. I just want to ensure that the print speed is not going to be to slow.

curieos, 2017-05-19:
I don't think this is a drastic change. If you don't want to use the feature, just set the min speed to 0. If it defaults to 0 then no one will be affected.

cerdeira, 2017-05-19:
As an option, yes, makes sense.