slic3r/Slic3r #3968: Does not send SLIC3R_ environment variables to python scripts

expertmm, 2017-05-19:
### Version
1.3.0-dev git 2017-05-09
from zip filename: slic3r-master.2017.05.08.1002.788cbeb

Operating System: Windows 10

### Behavior
* Does not send SLIC3R_ environment variables to python scripts. I am not sure if there is any way to get around this without writing a pearl script that passes the variables as arguments to python.
* Steps to Reproduce: Install python 3.4 64-bit from python.org, running as administrator, installing to C:\Python34, and adding it to the path. Associate py files to python.exe and pyw files to pythonw.exe. In Slic3r, add post-processing script to Print settings, Output options: choose a python script such as from the download below.
[testvars.zip](http://www.axlemedia.net/downloads/testvars.zip "testvars.zip")
* Expected Results: finds SLIC3R_ variables
* Actual Results: finds 0 environment variables with name containing SLIC3R_ (though works fine with pearl scripts, does not work with python scripts)

alexrj, 2017-05-20:
Can you please try a more fresh dev build? Yesterday I merged some changes which affect how environment variables are set on Windows.

However, I don't see why would a Perl script read environment variables and a Python one would not. Environment variables are passed by the operating system, and there's no correlation between the programming languages the two processes are written in (and Slic3r sets environment variables in C++, not Perl).

expertmm, 2017-05-22:
Wow! After updating to slic3r-master.2017.05.21.1096.450a918.32bit.zip, the script works perfectly now.
(says "Found 87 SLIC3R_ environment variable(s)"). Thanks.

closed 2017-05-22 by expertmm

alexrj, 2017-05-22:
Perfect, thanks for confirming this!