slic3r/Slic3r #3969: Recent (currently latest) git fails to build.

rewolff, 2017-05-21:
### Version
c4fe123b321be59cc5ed7c7d382bfefde64a19fd

### Operating system type + version
Ubuntu 16.04.2 AMD64. 

### Behavior
Fails to build with: 
```
In file included from src/libslic3r/Layer.hpp:5:0,
                 from src/libslic3r/Model.hpp:7,
                 from src/libslic3r/IO.hpp:5,
                 from src/libslic3r/IO.cpp:1:
src/libslic3r/Flow.hpp:14:6: error: use of enum ‘FlowRole’ without previous declaration

```
as the first error. While version b7544e3c6f32eace65b7bd736cfb6d5f1d664b16 does build.

alexrj, 2017-05-21:
Thank you for that, it was a typo left from some documentation work. Fixed now.

closed 2017-05-21 by alexrj

lordofhyphens, 2017-05-21:
Yeah, sorry about that, I was already half asleep when I saw the build
failures.

Thanks to @alexrj for sorting it out.