slic3r/Slic3r #3970: no way to close settings window.

rewolff, 2017-05-21:
### Version
b7544e3c6f32eace65b7bd736cfb6d5f1d664b16

### Operating system type + version
Ubuntu 16.04.2

### Behavior
I much prefer the tabbed configuration of the settings (like in 1.2.9) as opposed to the windowed that it is now. That's a personal preference. 

But what /is/ a problem is that on my window manager, the popup comes up without a close button. Apparently it's declared as some popup where I can normally click somewhere inside to close that popup. I'm guessing that the developer who made this has a window manager that does not listen to the "please do not provide a close button" that is passed to the window manager.

alexrj, 2017-05-21:
On the contrary, the dialog is created with `wxDEFAULT_DIALOG_STYLE` which implies `wxCLOSE_BOX`, so if the close button doesn't show it's a bug of the wxWidgets implementation for your window manager.
Can you provide a screenshot and more information about your setup? I can confirm the close button is there using a vanilla Ubuntu 16.04.

The dialog can also be closed by pressing the Esc key.

rewolff, 2017-05-21:
I'm using "sawfish" as my window manager. (it provides a feature I got addicted to early ninety nineties.)

That escape key is a convenient shortcut :-)

Screenshot: 
[image]

NateTG, 2017-05-22:
You can try hitting escape.

rewolff, 2017-05-22:
The "help->about"  dialog has even less reason for internal buttons like "close", "ok, I'm done with this". 

To me it's at least inconsistent. Most other programs do have an "ok/apply" button. Does the slicing start immediately when I change a parameter? With this new popup window it makes more sense to wait for the "OK" from the user (or possibly: "Cancel: I messed up, please return me to the old settings!".)

foreachthing, 2017-05-23:
The same is for the tabbed version (1.3.0 dev); the Print Settings tab has no X to close it with.

Lenbok, 2017-05-23:
@foreachthing it's even weirder - it seems to be whichever is the first tab that you open that is missing the X to close. If you open the Filament Settings tab first it's missing the X and then Print Settings does get the X.

foreachthing, 2017-05-23:
@Lenbok you're right! :-) I didn't even notice since I rarely use the other settings anymore.
@lordofhyphens how is that not a bug?

rewolff, 2017-05-23:
Ehh. For my understanding.... Some others are also seeing the "no close button" issue that I'm seeing?
On my system (Linux with sawfish as the windowmanager, all settings popups remain close-X-less...

lordofhyphens, 2017-05-23:
@rewolff it depends on your WM styling apparently. Which is probably not what we want in all cases.

rewolff, 2017-05-23:
Well, agreed. But if I'm the only one still using sawfish, and sawfish ignores the 'please render a close button' hint from the application, I can understand the: "it's not a bug on our side, it's in sawfish". 
But if others are seeing this on e.g. Windows, then I'd say that adding an "OK" button in the dialog is a good idea. And the hint that that's the case is the "help->about" popup that also has an "OK" button that makes the window go away.....

lordofhyphens, 2017-06-04:
[image]

lordofhyphens, 2017-06-04:
http://docs.wxwidgets.org/trunk/classwx_dialog.html

According to this, there should be a close box on the frame (meaning the outer frame, not a button in the bottom corner). 

You can do a little bit of testing via the following:
* Replace line 4 in ./lib/Slic3r/GUI/PresetEditorDialog.pm with:
````
use Wx qw(:dialog :id :misc :sizer :button :icon wxTheApp WXK_ESCAPE wxCLOSE_BOX);
````
* Replace line 14 with:
````
        wxCLOSE_BOX | wxDEFAULT_DIALOG_STYLE | wxRESIZE_BORDER | wxDIALOG_EX_METAL);
````

If you still aren't getting a close box, then it's very likely an issue with sawfish. If you do, it's certainly an issue with your WM or how wxWidgets is implemented (if you're building from source).

Windows 10 is just fine, as can be seen above.

rewolff, 2017-06-04:
I'm using the default "Crux" theme in swafish. A quick scan of some themes shows that most of them behave just like Crux, but some (Elberg, Hacksaw) render the close button on the settings frame. 
I was unable to find out where I'd be able to change this. 

Still, apparently many theme-writers have decided NOT to render a close button on this type of window. Probably because 100% of the tested applications didn't need one (= rendered a close button inside the frame).

lordofhyphens, 2017-12-30:
@rewolff since this is both upstream and you can make it a tab again, can I close this?

closed 2018-01-01 by rewolff