slic3r/Slic3r #3971: Build only uses one CPU.

rewolff, 2017-05-21:
### Version
fbb606d026bcf2e6825306bb49c5691f742bc464

### Operating system type + version
Ubuntu 16.04.2
### Behavior
Compilation takes longer than necessary: Only one core is being used in the compilation process. I'm using the documented procedure for Linux. 
If the build step was "make" then I'd tag on "-j 5" to try to keep  my four cores active. (the fifth job uses the CPU when one of the others needs to go to the disk). From how my system is behaving, I expect  a 4 fold increase in speed from the -j 5...

lordofhyphens, 2017-05-21:
Not fixable until we change the build system and finish shucking the perl.

rewolff, 2017-05-21:
:-(

lordofhyphens, 2017-05-21:
If you want to help, start porting more of the UI to C++ via the xsgui branch.