slic3r/Slic3r #3972: Add an option for extruder to heat the extruder anyway even if it's not used?

thuSkywalker, 2017-05-21:
### Version
slic3r-master.2017.05.17.1053.5f6214b.64bit

### Operating system type + version
Win10 64bit

### Behavior
Dual extruder printer, similar to Makerbot Replicator 1 dual.
If I let Slic3r use only one extruder, another won't heat. i.e. the temperature of another extruder is not set in G-code.
A cold extruder with filament leaked out of the nozzle then became hard would cause problems.

Would it be better if the behavior is like this:
In "Printer Settings", add an option "Heat the extruder even if not used".
For extruders with this option enabled, set its temperature according to the material assigned to it whether it actually extrude anything during the print.
For extruders with this option disabled, keep the behavior unchanged. i.e. its temperature is not set in G-code if it doesn't extrude anything during the print.

lordofhyphens, 2017-05-21:
Isn't this what start gcode is for?

alexrj, 2017-05-21:
You can just add the relevant temperature commands in your Start G-code.

lordofhyphens, 2017-05-21:
http://manual.slic3r.org/simple-mode/simple-mode#printer-settings see the bottom of the page.

uclaros, 2017-05-22:
Oozing can be an issue too with an unused heated extruder.

thuSkywalker, 2017-05-25:
Ooops I didn't know I can write something like this in start Gcode
`M104 S[first_layer_temperature_0] T0`

Ignore me.....I think the "feature" I proposed would add unnecessary complexity to the code.

closed 2017-05-25 by thuSkywalker