slic3r/Slic3r #3976: Question about multiple extruders and "default extruder"

mron, 2017-05-22:
### Version
1.29 and 1.3 beta

os x and linux 14.04
### Behavior
What is the "default extruder" ? What is it set to when I set Print Setting -> Extruder -> ... from 1 to 2?
Is "2" now the default extruder? It doesn't seem to be.
When printing with multiple materials for one object I find I need to set:
Print Settings -> multiple extruders to the values I want.
Then for each object that needs extruder 2 I have to set:
Object Settings -> Extruder
and set Object Settings -> Extruder -> Infill
and set Object Settings -> Extruder -> Perimeter
and set Object Settings -> Extruder -> Solid Infill

This is a lot of setting. I would like Perimeter and Solid Infill to be set to Infill extruder unless explicitly set.

An area of confusion for me is when setting extruders in the Object Settings what is "default extruder"? I expect when I set Print Settings -> Extruder -> Infill 
and Print Settings Settings -> Extruder -> Perimeter,
and Print Settings  Settings -> Extruder -> Solid Infill that these are the default and any additional parts loaded with the object and set to "default" would use it. This doesn't seem to be the case.

Maybe "default" should just be eliminated and extruders should be explicitly set.
I apologize for the poor wording.

alexrj, 2017-05-29:
"Default" means the object will use the extruders set in *Print Settings > Extruders*.

Your suggestion to remove it does not make much sense and would require users to always set extruders for each single object. Why such sadism?

mron, 2017-05-30:
Thanks for replying,
I agree with reducing clicks! That's kind of my point. When I use multiple objects with multiple materials, the object that uses the extruder from "Print Settings" seems to be fine. The object that uses the other extruder need to have the extruder set in 4 places:
Object Settings -> Extruder
and set Object Settings -> Extruder -> Infill
and set Object Settings -> Extruder -> Perimeter
and set Object Settings -> Extruder -> Solid Infill

This is a lot of clicks and error prone. I would like to simply set the "Object Settings -> Extruder" and have the other 3 follow along, unless I choose otherwise.

I'll come up with a simple test to show what I mean.