slic3r/Slic3r #3977: Unreadable text while printing via USB - white on white

7eggert, 2017-05-22:
### Version
1.3.0 dev 6b68b71-linux-x64 (binary download)

### Behavior
The picture says it all

[image]

lordofhyphens, 2017-05-22:
Looks like a strange interaction with your theme.

7eggert, 2017-05-22:
Yes, strangely it's not just enough to set one color (foreground/background) to have the other magically match :-)

lordofhyphens, 2017-05-23:
I didn't realize that @alexrj set any colors at all there.

alexrj, 2017-05-29:
I should have fixed this, but can't test. @7eggert, do provide feedback!

7eggert, 2017-05-29:
7eggert@be10:~/.lib/slic3r-1.3.0-dev-c22d20b-linux-x64$ ./Slic3r --gui
Xlib:  extension "RANDR" missing on display ":0".
Can't locate loadable object for module Wx::AUI in @INC (@INC contains: /home/7eggert/.lib/slic3r-1.3.0-dev-c22d20b-linux-x64/local-lib/lib/perl5/5.24.0/x86_64-linux-thread-multi /home/7eggert/.lib/slic3r-1.3.0-dev-c22d20b-linux-x64/local-lib/lib/perl5/5.24.0 /home/7eggert/.lib/slic3r-1.3.0-dev-c22d20b-linux-x64/local-lib/lib/perl5/x86_64-linux-thread-multi /home/7eggert/.lib/slic3r-1.3.0-dev-c22d20b-linux-x64/local-lib/lib/perl5 /home/7eggert/.lib/slic3r-1.3.0-dev-c22d20b-linux-x64/lib ./local-lib/lib/perl5/x86_64-linux-thread-multi ./local-lib/lib/perl5 /home/7eggert/.lib/perl /home/travis/perl5/perlbrew/perls/slic3r-perl/lib/site_perl/5.24.0/x86_64-linux-thread-multi /home/travis/perl5/perlbrew/perls/slic3r-perl/lib/site_perl/5.24.0 /home/travis/perl5/perlbrew/perls/slic3r-perl/lib/5.24.0/x86_64-linux-thread-multi /home/travis/perl5/perlbrew/perls/slic3r-perl/lib/5.24.0 .) at /home/7eggert/.lib/slic3r-1.3.0-dev-c22d20b-linux-x64/local-lib/lib/perl5/x86_64-linux-thread-multi/Wx/Mini.pm line 78.
Compilation failed in require at /home/7eggert/.lib/slic3r-1.3.0-dev-c22d20b-linux-x64/local-lib/lib/perl5/Slic3r/GUI/MainFrame.pm line 13.
BEGIN failed--compilation aborted at /home/7eggert/.lib/slic3r-1.3.0-dev-c22d20b-linux-x64/local-lib/lib/perl5/Slic3r/GUI/MainFrame.pm line 13.
Compilation failed in require at /home/7eggert/.lib/slic3r-1.3.0-dev-c22d20b-linux-x64/local-lib/lib/perl5/Slic3r/GUI.pm line 33.
BEGIN failed--compilation aborted at /home/7eggert/.lib/slic3r-1.3.0-dev-c22d20b-linux-x64/local-lib/lib/perl5/Slic3r/GUI.pm line 33.
Compilation failed in require at (eval 84) line 1.

lordofhyphens, 2017-06-04:
@7eggert That should be fixed now (was a packaging bug) as of 6a7057b1ebf09710fa6452013d20b8ea082d01e3. 

Please test!

lordofhyphens, 2017-06-04:
I just tested it on XFCE-dusk theme which has dark background and white text.

[image]

closed 2017-06-04 by lordofhyphens

7eggert, 2017-06-05:
Confirmed