slic3r/Slic3r #3978: Make use of .obj texture vertices?

mron, 2017-05-23:
### Version
1.3

### Operating system type + version
Any

### Behavior
Obj files support texture vertices. I would like to use texture vertex values to alter print behavior. 
One thing that comes to mind is to change the number of perimeters or extrusion width to create a visual texture.  #3750 seems to discuss a gcode post processing language. Maybe texture info could passed to this language.

Displacement mapping? Procedural textures?

_Is this a new feature request?_

Yes

I don't know how to tag an issue. This would be a feature request.

lordofhyphens, 2017-05-30:
This would need to wait, I think, for replacing admesh for vcglib; which is a very flexible system.

lordofhyphens, 2017-05-30:
I think you'd also need to describe the use case a bit better with pictures (and some test files); this isn't something I've thought about as being useful at all.

mron, 2017-05-31:
It's a bit blue sky-ish. I like to print thin single wall objects that light passes through. I have experience writing shaders, so to be able to vary extruding thickness, number of perimeters or extruder based on texture maps, UV coordinates and texture images, or a "shading" language would be fun and easier than trying to apply fine texture detail in a model. Extend the 2D textures to 3D, then varying infill based on volume functions could be interesting for both art and structural attributes.

It's out there, but someone will do it someday. Maybe just leave the hooks in place for someone to try.