slic3r/Slic3r #3981: Height skirt doesn't deseabled

Pydo44, 2017-05-28:
### Version
1.3.0-dev mac 10.11.5

### Behavior
I had put a skirt height at 2000 . like that i was certain to have a skirt for all the print
but it was completely usefull for a pure pla print.
so i change the config to 10 and the skirt stay for all the print.....* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Pydo44, 2017-05-30:
Hi today for a big shape i had an error message
[image]

Pydo44, 2017-05-31:
Ok i can do again the issue
with 1 object on the plater, i choose in the setting for this object the default extruder.
the skirt is at full height
now i choose in the setting of the object the extruder 1 ( all object and support are in the same material) but the skirt can be change in the preview

in the parameter i choose all for extruder 1 and only support/skirt/raft for extruder 2

and at the print the skirt is printed with the extruder 1.....

thanks all

Pydo44, 2017-06-03:
Eureka ..... I found the problem with the skirt height ....

when you check the ooze prevention option the skirt height is put at the maximum B<

alexrj, 2017-06-03:
Next time please follow the instructions. We need STL file and config, and proper steps to reproduce the issue. You didn't mention ooze prevention.

closed 2017-06-03 by alexrj

Pydo44, 2017-06-04:
sur i didn't mention it , i didn't know is was checked and ididn't konw how to reproduced the problem

so i warning you for the bug and try to give you the maximum information and you are shooting me..... pfffff
next time if you want a better tester give some money and we spend much time to do the research and correct the software

lordofhyphens, 2017-06-04:
@Pydo44 The issue description is useless to us without the information asked for in the text you deleted to write it.

Configuration, a model that shows the problem with configuration, and whatever else you did to make it show up. Without that information, we have nowhere to start and as such can do nothing.

Ooze prevention uses the skirt to purge unused extruders.

lordofhyphens, 2017-06-04:
@Pydo44 neither @alexrj nor I generally get paid for fixing issues like this-we do this in our spare time when we aren't at our real jobs.