slic3r/Slic3r #3984: Controller tab always visible

gege2b, 2017-05-29:
hi there

### Version
current head ( c22d20b9bb167796fedbcc3e9e458b0fb143366d )

### Operating system type + version
linux mint serena (kde)

### Behavior
* _Describe the problem_

since the mentioned commit, the controller tab is back and I'm unable to hide it (or I don't know how to do it)
I've nothing in printer settings -> USB/Serial connection

* _Steps needed to reproduce the problem_

just fire up slic3r
for testing purpose, I've deleted my slic3r.ini but it doesn't change anything

* _Expected Results_

controller tab to show if needed

* _Actual Results_

controller tab is here

  * _Screenshots from __*Slic3r*__ preview are preferred_

#### STL/Config (.ZIP) where problem occurs

default settings and no need to load a model

alexrj, 2017-05-30:
That option was merged by mistake. It also had a true default, so most users ignored that feature.
What's the problem of having the tab there? We have the tab bar anyway, so there's no extra space used. Just don't use controller if you don't need it :-)

gege2b, 2017-05-30:
Well, I agree that the presence of the controller tab is not really a problem.
I had it hidden because I don't use it, so when it just reappeared "by magic" with no (visible) way to hide it again, I though it was a bug or a regression, hence this report :)

BETLOG, 2017-09-28:
The fact that it's forced to be there, while the rest of the tabs that I actually *want* to be there all the time, and not just after I waste moves and clicks pulling them up.. is very annoying.
If it was simply present next to Print/Filament/Printer tabs it wouldn't be so bad, but when you don't ever use it via direct USB connection (use octoprint instead), AND the desirable tabs are absent unless invoked through a rather tedious series of actions... it's a bit of an in-my-face peeve.
Prusa version allows it to be disable din preferences dialog. Which is nice.

lordofhyphens, 2017-09-28:
Done. Leaving the controller tab set to visible to avoid people never finding the feature while allowing people for whom it annoys to turn it off.

commit 90f66ea9 (2017-09-28): Allows hiding of the Controller tab (controller tab still defaults to visible).

Fixes #3984
closed 2017-09-28 by lordofhyphens