slic3r/Slic3r #3985: Slic3r not starting on windows 10

ew1773, 2017-05-29:
### Version
Latest builds Slic3r-master.2017.05.29.1131.c22d20b and Slic3r-master.2017.05.26.1125.3469800

### Operating system type + version
Windows 10 Version 1703 Build 15063.296

### Behavior
slic3r.exe and slic3r-console.exe fail to start and deliver the following error.

[image]

perl524.dll does exist in the libexec folder.

Both files were downloaded and then unzipped into a clean folder. I used the default windows tool and 7zip.

Let me know if more information is needed or I missed something obvious. 

_Is this a new feature request?_
No

lordofhyphens, 2017-05-30:
I noticed a regression in the packaging;  the architecture (64bit or 32bit) should have been in the name. I've fixed that and will be testing when the build server is done.

I also am wondering if the redirect wasn't defined properly and I just didn't test it right on my dev system.

ew1773, 2017-05-30:
Let me know when you have it and I can test on my machine too.

lordofhyphens, 2017-05-30:
It looks like it's related to the changes that @alexrj made to clean up the host directory; the library path for the exec isn't looking in the libexec dir. I'll see what I can do about this.

lordofhyphens, 2017-05-30:
As a workaround, you can get an exe from an older build (before everything was moved into libexec) and copy it into the libexec dir (run from there).

Slic3r.exe is just a thin wrapper shell that executes the perl script, so there is no harm in this.

lordofhyphens, 2017-05-30:
The other workaround is to install Slic3r perl, which is why it looked like it worked on my development system. I've got a reproduction case here so currently trying to work out how to get the desired behavior.

lordofhyphens, 2017-05-30:
Alright, I've reverted the packaging changes that caused the problem; the organization that @alexrj is interested in doesn't really work on Windows. It's theoretically possible on Windows 8 or higher and Windows 7 with a specific patch installed. 

I've reverted the branch that got merged in for now.

ew1773, 2017-05-30:
I retried these builds:
Slic3r-master.2017.05.30.1133.1be266c.64bit and Slic3r-master.2017.05.30.1133.1be266c.32bit
They gave the same error. Let me know if i can do anything to help.

lordofhyphens, 2017-06-03:
@ew1773 this should work now for you.

closed 2017-06-03 by lordofhyphens