slic3r/Slic3r #3986: Unable to launch Slic3r for MacOS 10.12.5

mrmacnology, 2017-05-30:
### Version
slic3r 1.3.0 dev 8030b00

### Operating system type + version
MacOS 10.12.5

### Behavior
Once installed and the developer is approved Slic3r fails to open. I double click to open it, the icon jumps on the dock and then disappears.

I saw this reported on Jan 26 #3689 but I was unable to successfully open the dev version.

Thank you for your help. I'm new to Slic3r.

zargony, 2017-05-30:
Same here. Launching it from terminal shows the error message:

```
~$ /Applications/Slic3r.app/Contents/MacOS/Slic3r --gui
Can't locate loadable object for module Wx::AUI in @INC (@INC contains: /Applications/Slic3r.app/Contents/MacOS/local-lib/lib/perl5/5.24.0/darwin-thread-multi-2level /Applications/Slic3r.app/Contents/MacOS/local-lib/lib/perl5/5.24.0 /Applications/Slic3r.app/Contents/MacOS/local-lib/lib/perl5/darwin-thread-multi-2level /Applications/Slic3r.app/Contents/MacOS/local-lib/lib/perl5 /Applications/Slic3r.app/Contents/MacOS/lib /Applications/Slic3r.app/Contents/MacOS/local-lib/lib/perl5/darwin-thread-multi-2level /Applications/Slic3r.app/Contents/MacOS/local-lib/lib/perl5 /Users/buildbot/perl5/perlbrew/perls/perl-5.24.0_osx10.7/lib/site_perl/5.24.0/darwin-thread-multi-2level /Users/buildbot/perl5/perlbrew/perls/perl-5.24.0_osx10.7/lib/site_perl/5.24.0 /Users/buildbot/perl5/perlbrew/perls/perl-5.24.0_osx10.7/lib/5.24.0/darwin-thread-multi-2level /Users/buildbot/perl5/perlbrew/perls/perl-5.24.0_osx10.7/lib/5.24.0 .) at /Applications/Slic3r.app/Contents/MacOS/local-lib/lib/perl5/darwin-thread-multi-2level/Wx/Mini.pm line 78.
Compilation failed in require at /Applications/Slic3r.app/Contents/MacOS/local-lib/lib/perl5/Slic3r/GUI/MainFrame.pm line 13.
BEGIN failed--compilation aborted at /Applications/Slic3r.app/Contents/MacOS/local-lib/lib/perl5/Slic3r/GUI/MainFrame.pm line 13.
Compilation failed in require at /Applications/Slic3r.app/Contents/MacOS/local-lib/lib/perl5/Slic3r/GUI.pm line 33.
BEGIN failed--compilation aborted at /Applications/Slic3r.app/Contents/MacOS/local-lib/lib/perl5/Slic3r/GUI.pm line 33.
Compilation failed in require at (eval 84) line 1.
```

lordofhyphens, 2017-05-30:
https://github.com/alexrj/Slic3r/blob/master/package/osx/make_dmg.sh#L135

That line in the packaging removes the reference to wx aui, which is used now as of 060187ad1703f70c8e2c2847829216c38e889403.

It worked in testing because our dev systems don't have that library removed.

lordofhyphens, 2017-05-30:
I pushed a small change to the packaging script that leaves AUI in. Next build should have the fix in it.

@alexrj if you added aui in error feel free to fix it the other way around. I can't do involved edits from my phone ;)

lordofhyphens, 2017-05-30:
@mrmacnology  @zargony when a build gets done in the next hour or so, please test so I can close this issue.

mrmacnology, 2017-05-30:
Thank you @lordofhyphens. Will do.

lordofhyphens, 2017-05-30:
Looks like it's ready now.

mrmacnology, 2017-05-30:
Slic3r launched with 1.3.0 dev C837A06. When I get home I will run through the configuration assistant. Thank you for your help!

closed 2017-05-30 by lordofhyphens

zargony, 2017-05-30:
Works fine here, too. Thanks everybody!