slic3r/Slic3r #3987: How compile slic3r as a single .exe in windows and let it work correctly?

shushengyu, 2017-05-30:
### Version

Slic3r-cleanup-windows
### Operating system type + version
windows10,boost-1.6.3.0

### Behavior
I have compiled the soruce by followint the instructions in windowss-build.txt with CMake-Gui,
Before I successed I have to do some repairs. 
I have get the .exe file:extrude-tin.exe and slic3r.exe, but it seems wrong.
For example, I used it to convert a .stl file to a .svg file like this:
D:\WORK\3dprinter\slic3r-study\Slic3r-cleanup-windows\build>**slic3r D:/WORK/3dprinter/Stanford_Bunny/bunny.stl --layer-height 0.2 --output D:/WORK/3dprinter/Stanford_Bunny/output3.svg --export-svg**
SVG file exported to D:/WORK/3dprinter/Stanford_Bunny/output3.svg

But the result is sad,the .svg file like this:
```
<!-- Generated using Slic3r 1.3.0-dev http://slic3r.org/ -->
	<g id="layer0" slic3r:z="0.0000" slic3r:slice-z="0.0000" slic3r:layer-height="0.0000">
	</g>
	<g id="layer1" slic3r:z="0.2000" slic3r:slice-z="0.1000" slic3r:layer-height="0.2000">
      ...(all are same)...
<g id="layer427" slic3r:z="85.3998" slic3r:slice-z="85.2998" slic3r:layer-height="0.2000">
	</g>
	<g id="layer428" slic3r:z="85.5998" slic3r:slice-z="85.4998" slic3r:layer-height="0.2000">
		<path d="M 42.5077 23.88 42.607 23.7619 42.8973 23.3403 43.0802 23.0657 43.1399 22.4314 43.2028 22.2493 43.0508 22.2313 43.0013 22.3064 42.8855 22.3855 42.5956 23.0446 42.2028 23.4336 42.1648 23.6091 42.418 24.0158 z " style="fill: white; stroke: black; stroke-width: 0; fill-type: evenodd" slic3r:area="-0.0021" />
	</g>
```
In fact, the official release stale version will give the result like this:
 ```
<g id="layer0" slic3r:z="1.75e-007">
    <polygon slic3r:type="contour" points="23.104864,21.940072 23.25317,23.224072 23.212769,23.256456 22.973736,23.17747 22.842856,23.81692 22.56023,23.797806 22.451631,22.854028 22.281996,22.388648 22.577888,22.352518 22.829012,21.790487 22.898389,21.634682 23.03505,21.328948 23.757988,21.17604" style="fill: white" />
  </g>
  <g id="layer1" slic3r:z="4.5e-007">
    <polygon 
     .......................................................
```
 I have tryied many versions, all the result are same. what's the wong about this? How does the perl package start the core of slic3r.exe? Is there lacked some configurations?
  I need some help, thank you very much!
 
#### STL/Config (.ZIP) where problem occurs
any

lordofhyphens, 2017-05-30:
The slic3r.exe that goes out with our development versions is a thin wrapper to run the perl script. It's unnecessary if you have slic3r-perl installed (but helpful when dealing with other programs that expect an executable). Its source is in `package/common/shell.cpp` and is built by our build server with `package/win/compile_wrapper.ps1`. 

The one built from the `src` directory (using cmake) is a CLI-only slicer (no perl components). I don't have much experience with it. 

Why are you using Slic3r-cleanup-windows? That's an experimental branch and shouldn't be used unless you know exactly why you want it (I just deleted the branch as well, it doesn't work for its stated purpose). 

I just updated the wiki: https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows#building-slic3r

lordofhyphens, 2017-06-01:
https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows#compiling-slic3r-cli-only-limited---no-perl for building the alternative cli (no perl) Slic3r.

lordofhyphens, 2017-06-01:
For the perl-dependent Slic3r, there is no way to package the entire application into one exe file that doesn't have many tradeoffs. 

If you really really want to go forward with this, examine the PAR package and pp and package_win32.ps1

closed 2017-06-01 by lordofhyphens