slic3r/Slic3r #3989: Slic3r fails to generate support material for horizontal overhangs (with example)

ulfgri, 2017-05-30:
[Support_failure_example.zip](https://github.com/alexrj/Slic3r/files/1039730/Support_failure_example.zip)
### Version
1.3.0-dev

### Operating system type + version
Linux boar 4.4.0-71-generic #92~14.04.1-Ubuntu SMP Fri Mar 24 15:22:50 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

### Behavior
No support material is generated by Slic3r for horizontal overhangs, no matter what the setting for the critical angle is. Confirmed this problem with several models. A simple example etc. is attached.

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Other than that, Slic3r is fantastic. Thank you !

mrvanes, 2017-05-31:
Just out of curiousity I looked at your file and was surprised that it didn't generate support, until I unchecked the "Don't support bridges" option. So, the overhang is detected as a bridge I guess?

ulfgri, 2017-05-31:
Hi Martin,
      
      you are right, when I unclick "Don't support bridges" then
      everything works as expected. Didn't occur to me to unclick this
      option.
      
      Ulf
      
      On 05/31/2017 05:57 AM, Martin wrote:
    
    
      Just out of curiousity I looked at your file and was surprised
        that it didn't generate support, until I unchecked the "Don't
        support bridges" option. So, the overhang is detected as a
        bridge I guess?
      —
      
        
          
          
        
        
      
    
    
    
Ulf Griesmann               \\  tel: +1 301 339 4962
16833 Westbourne Terrace     \\ 
Gaithersburg, MD 20878-2033   \\  e-mail: ulfgri@gmail.com

RichyT, 2017-06-05:
I wonder if this is my issue. I am trying to print a dome. It does generate support but it was generated at more like 60 degrees than 45.