slic3r/Slic3r #3990: Dual Extruder - Print ONLY with Extruder 2 - how???

imaluschi, 2017-05-31:
### Version
Slic3r 1.3.0-dev
Downloaded and built on Don 18 Mai 2017

### Operating system type + version
Linux, Debian Stretch

### Behavior
Have built up dual extrusion on my Ultimaker Original +. As Cura can't do dual extrusion, I'm trying to do this with Slic3r. Didn't try any dual extrusion, first every extruder should work fine, then I'll try further...

Extruder 1 works nice, so now I want to test extruder 2. But I can't get a print with only extruder 2. Maybe someone can help me.
Settings: E1 - ABS, Nozzle 0.5mm; E2 - PLA, Nozzle 0.3mm (see screenshots).
In print settings all is set to extruder 2. But when i export the G-Code file and try to print following behaviour:
BedTemp is set for First Layer of E1 (means 110°C) instead of 60°C for E2 (PLA).
Manually setting down Temp. E1 starts the printing, but E2 is not on Temp. At one try the printer shut down because Bed, E1 and E2 was heating - too high power for the system.

So I tried to change the G-Code file, but I'm not very used to it. Result was heating bed to right Temp. heating E1 (wrong extruder) to right Temp. (215°C). Changed it manually, but when the print started the extruder motor 2 didn't move, even with E2 on right Temp.

So I don't get it. If I define only E2, why is E1 heated up? If I only use E2, why is the Bed heated to FirstLayerTemp. of E1?
Custom G-Code for the Temperatures:
```
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
```

[image]
[image]
[image]
[image]
[image]

Anyone an Idea?

lordofhyphens, 2017-05-31:
Quick solution is to create a second Printer profile that has 1 extruder in it, and to add a "T1" command in your start gcode (to switch to that extruder).

That most closely resembles what you are trying to do.

mron, 2017-05-31:
The quick solution is the only solution I think. This relates to #3976. I don't know how to get Slic3r to consistently output a tool gcode by relying on the "default extruder".

imaluschi, 2017-05-31:
Thanks for your reply, I'll try it when time allows ;-)

Pydo44, 2017-05-31:
i think select the 3d view click on the object for contextual menu 
select setting and choose your extruder for this object