slic3r/Slic3r #3992: speed for the second extruder

Pydo44, 2017-05-31:
### Version
Slic3r 1.3.0 -dev

### Behavior
on the second extruder i have another kind of material* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2017-05-31:
For the same kind of moves?

Pydo44, 2017-05-31:
I don’t understand

for the first extruder i have some PLA and the second some PVA or ABS

i need to have for each extruders some different speed  for all the parameter (infill, perimeter,…..)

lordofhyphens, 2017-05-31:
Currently you can:

Assign an extruder per-object in the object settings right-click menu. You
can also override speeds here.

Or you can assign extruders on a per - role basis. In this case, change the
speed of the associated print move.

What are you doing differently?

Pydo44, 2017-05-31:
so i didn't see the override in the setting....
but in this case i need for each object modify the setting..... B<

and your second way, i can't understood what you say,

but i think the configuration of the parameters for the different speed must be in each filament instead of print settings 
like that we can print with a filament (for the support for exemple) at lower speed 
or print with the same filament, all the obect at lower speed (perimeters, infill....)

and with this feature we can have on our printer two differents filaments (PLA and ABS) and print easily with the first or second extruder

lordofhyphens, 2017-05-31:
There are a number of roles (infill, perimeter, etc) that you can assign different extruders to in Print Settings (do infill with one extruder, perimeters with another, support with a third, etc). 

The rest of your statement leads me to believe that: 
* You have a general idea of "how much faster" or "how much slower" in printing in PETG is vs printing in PLA (or ABS). 
* You'd like to be able to capture these differences with the filament settings so you can either mix two or more filaments and print both well with the same print profile.

The underlying goal here is to reduce the need for multiple profiles to tweak speed and to permit multiple extruders to use the same print profile (assuming the only thing that needed to change was speed, of course). 

Something like defining speeds relative to the currently defined print profile speed with an additional threshold value (for minimum speed, etc). We might be able to get away with two values (% and minimum speed to not reduce if the origin speed is below that already) that affect all print moves, but I could see some reasons to break it out to all of the different print moves. 

I can see this not playing well with autospeed. Any thoughts @alexrj? 

I can see some related use cases here having to do with temperature and speed changes in the print as well, but that would take some further experimentation.

Pydo44, 2017-06-01:
thanks for the jobs