slic3r/Slic3r #3993: Internal brim for isles / supports

Leozolt, 2017-05-31:
Hello I
[Mould-top.zip](https://github.com/alexrj/Slic3r/files/1043040/Mould-top.zip)
 am using 1.3 dev and the internal brim feature comes handy for me but it doesn't work on internal "exteriors" / let's say supports that are internal.

here is a zipped stl as an example and the first layer as picture showing how the center doesn't have a brim if I set 5mm exterior and 5mm interior brim.
[image]

Regards TT

Leozolt, 2017-05-31:
I also understand that I should/could enlarge the center somehow but for me the issue is that ther is no brim on it.
Regards and thank you for understanding,

TT

lordofhyphens, 2017-06-01:
@Leozolt the config needs to be included, your text doesn't indicate if you provided an exported config or not.

Leozolt, 2017-06-01:
@lordofhyphens I have added the config in this zip. (config bundle crashes so this is a simple config export) It's not generated support material, it's a pillar inside the piece that acts as a support though.

Regards, TT
[Mould-top.zip](https://github.com/alexrj/Slic3r/files/1044611/Mould-top.zip)

lordofhyphens, 2017-06-01:
Thanks.

We want simple config exports in the first place. Bundles don't tell us
what you were using and contain a lot of unnecessary information.

Leozolt, 2017-06-04:
I think internal surfaces should also get brims.

What I mean is that isles inside an object (1 - 6) should get a brim on its outside/inside according to Exterior brim width/Interior brim width OR an optional "Inner isle exterior brim width/Inner isle interior brim width"(4 as example)

The inside of 4 should get a Brim anyway since it is an interior of the part (outer shell) itself?

[image]

[image]

This should be independent of #4004 that is more like post processing individual edges of surfaces.

regards TT