slic3r/Slic3r #3994: Concentric Infill on bottom layer starts from middle

roldershaw, 2017-05-31:
### Version
1.3.0-dev 1143 6a7056b x64

### Operating system type + version
Windows 7 Pro 64bit

### Behavior
Similar issue to https://github.com/alexrj/Slic3r/issues/898 
Gcode generated for concentric infill on the bottom layer starts from the middle rather than from the perimeter wall which causes poor adhesion. Works as expected in the stable version

Thanks for all your great work!

Example image
Dev version on left, latest stable on right

[image]

lordofhyphens, 2017-06-01:
Need a copy of a model that shows the behavior and config, as the text you deleted to write this issue says.

roldershaw, 2017-06-01:
Any chance I could email that to you? as the model belongs to a client I can't release it publicly.

lordofhyphens, 2017-06-01:
It just needs to be a model where you see the problem, not necessarily the one above.

Try doing a cut along Z close to the base and see if it still reproduces.

lordofhyphens, 2017-06-01:
The bottom mm or so of the model (don't generate top layers) would probably be sufficient.

Something else: there isn't any notices about repair, right?

roldershaw, 2017-06-01:
Hi,

Thanks for getting back to me. I've tested with a calibration cube and have reproduced the same issue. I've attached the model and config below.

The original model has no auto repair notices.

I have just had a little odd issue. If I try to export a config bundle from the dev version Slic3r crashes, I did originally import my 1.2.9 config bundle into the dev version which I have included in the zip also.

[3994_issue.zip](https://github.com/alexrj/Slic3r/files/1045630/3994_issue.zip)

lordofhyphens, 2017-06-01:
We know about config bundles crashing. We do NOT want you to upload config
bundles (use export config, as the prompt text indicates).

roldershaw, 2017-06-01:
That's fine, I uploaded a normal config from the dev version.