slic3r/Slic3r #3996: Crash on "Export Config Bundle

markolo25, 2017-06-01:
### Version
1.3.0-dev 6a7057b

### Operating system type + version
Windows 10 Version 1703 (OS Build 15063.296)
### Behavior
When exporting a config bundle the application closes
To Recreate the problem
File > Export Config Bundle > Save > "Application closes on it's own"

using slic3r-debug console 

[image]

lordofhyphens, 2017-06-01:
I suspect that there's a regression from the changes in gui-shortcuts (iterating over all of the configurations). 

Single-config export still works.

lordofhyphens, 2017-06-04:
Preset::load_config returns immediately if its internal config isn't empty. However, it returns nothing instead of `$self->_config`. 

Since this seems to be one of the only places that expects an immediate return and the normal behavior for Preset::load_config is to return a copy of the config object, we can safely return `$self->_config` during the early abort logic.

commit 1d6170bc (2017-06-04): Make Preset::load_config return a config if it short-circuits. Fixes #3996
closed 2017-06-04 by lordofhyphens