slic3r/Slic3r #3998: FR: Be able to set auto arrange offset

SturmGhost, 2017-06-01:
### Version

1.3.0-dev

### Operating system type + version

Windows 10 64 Bit

### Behavior

Auto arrange is quite annoying when dealing with different parts. Slic3r set's a really big offset between each part. Would be nice to be able to enter an offset by myself. 

This is crap (auto arrange): 

[image]

This is better (by hand):

[image]

foreachthing, 2017-06-02:
See here: https://github.com/alexrj/Slic3r/issues/3452#issuecomment-304854008

SturmGhost, 2017-06-02:
Okay but if you can arrange the parts in Slic3r then you should be able to control the parts offset... There is no need for 3. party software. Just give us an option for the offset. :)

lordofhyphens, 2017-06-02:
The minimum distance is currently exposed via a configuration option (it probably should be refactored out to the UI instead; doesn't need to be in libslic3r). 

The routine that does the arrange appears to live at https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/Model.cpp#L283 and called from https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/GUI/Plater.pm#L1314

So if someone wants to implement a better packing algorithm then knock yourself out.

If you just want to tweak `min_object_distance` then you can add it in https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/GUI/PresetEditor.pm (even to a compiled build, look in the `lib` directory) by editing PresetEditor.pm and using an existing option as a guide. 

See how that works for you and if it's great then we can look into either adding the option directly or refactoring it appropriately.

lordofhyphens, 2017-06-04:
This is also beyond setting a simple offset, getting something that looks like what you did by hand requires a better algorithm (and optimal algorithms for packing are usually NP-hard).

lordofhyphens, 2017-06-04:
Interested parties in making auto-arrange behave better should look at #3237.