slic3r/Slic3r #4000: Post-Processing Script Parameters?

foreachthing, 2017-06-02:
### Version
Any, so far.

### Operating system type + version
Windows 7 Pro

### Behavior
I made a post-processing script which accepts parameters (like: `Slic3rPostProcessing.exe --output "c:\GeeeeCode\[input_filename_base].gcode" --v 4 --i` ) but for some reason, whenever I add such a parameter to the script input box it will fail to execute my script.

```---------------------------
Error
---------------------------
The configured post-processing script is not executable: check permissions.
(c:\temp\Slic3rPostProcessing.exe --v 4 --i )
---------------------------
OK   
---------------------------
```

Is it not possible to pass parameters?

thuSkywalker, 2017-06-02:
I met the problem too. My solution is to write a `.bat` file, e.g. `postprocessing.bat` and Slic3r will run the script with filename as the first parameter: `postprocessing.bat <input_filename>`. Then you can pass the filename to `Slic3rPostProcessing.exe` in `postprocessing.bat`. See my configuration: https://github.com/thuSkywalker/wiiboox-c300-slic3r-config/blob/master/gpx/gpx.bat

foreachthing, 2017-06-02:
@thuSkywalker thank you! :-)

alranel, 2018-11-18:
Actually, that limitation was in place on purpose, for security reasons. I didn't want people to be able to distribute a malicious config file containing, say, `rm -rf ~/*` thus executing arbitrary commands on other people's machines.  Passing only the name of a single executable mitigates this risk. In order to supply arguments to a post-processing script, people could just create a wrapper shell script.

alranel, 2018-11-20:
I wonder if we should add an option to enable this, and leave it disabled by default.

foreachthing, 2018-11-20:
I understand your concern. But do you have a means to pass parameters the save way?
As long as one can execute something, there's a risk of misuse. I _could_ add that malicious command to my post-processor...

nwneisen, 2019-08-29:
It looks like #4363 added the ability to pass parameters to a post-processing script.  Is this issue still open because of the enable/disable feature?

foreachthing, 2019-08-29:
Nah, I forgot about it and never closed it ... we're done here!

closed 2019-08-29 by foreachthing