slic3r/Slic3r #4002: 3D rendering issues on Fedora 26

hroncok, 2017-06-02:
### Version
1.2.9

### Operating system type + version
Linux, Fedora 26 64bit (didn't happen on Fedora 25)

### Behavior
The 3D preview misses the plate ad have some extra lines.
[image]

It only happens with the Fedora package. ten binaries form slic3r.org do not exhibit the issue:
[image]

I was thinking maybe it is a dependency problem, but Prusa's Slic3r (Fedora package as well) does not exhibit the problem either.
[image]

I know you might say "the binaries provided work" and close this, but could you please help me investigate what's causing this? Does it look like something you've already seen? Thanks for help.

(Happens with any STL, any config.)

My GL related packages and their versions:

 * perl-OpenGL-0.7000 (was 0.6704 on Fedora 25)
 * perl-Wx-GLCanvas-0.09 (didn't change)

lordofhyphens, 2017-06-02:
Try installing the Perl local-lib package and installing a copy from
source. It won't hurt your existing system.

See if the behavior exists on that build.

In either case you should contact the Fedora package maintainer.

lordofhyphens, 2017-06-02:
There were some changes to how the 3d rendering was set up to deal with
opengl changes. The prusa3d branch and -dev have these changes, 1.2.9 would
not.

bubnikv, 2017-06-02:
@hroncok is the Fedora maintainer :-)

lordofhyphens, 2017-06-02:
This was a crash issue when it showed up for us #3575 and #3540 by the way, so you're probably ahead of the game ;)

hroncok, 2017-06-02:
#3575 fixes it, thanks. 
[image]

closed 2017-06-02 by hroncok