slic3r/Slic3r #4003: Slic3r should not add T0 argument to M104 command if number of extruders is set to 1

DrLex0, 2017-06-02:
### Version
1.3.0-dev-bcd8603 (latest dev build as of this writing)
(also present in Prusa3D fork)

### Operating system type + version
Mac OS X 10.11.6

### Behavior
* Even when using a printer profile with a single extruder, Slic3r still appends an extruder argument to the M104 command (at least for Sailfish firmware, haven't tested others). It must omit this argument if the number of extruders is set to 1. Otherwise, the wrong extruder can heat up when doing a single-extruder print on a multi-extruder printer.
* _Steps needed to reproduce the problem_
  1. Create a printer profile with the number of extruders set to 1 (or just load the attached config).
  2. Select a filament type that has different temperatures set for first and second layer.
  3. Slice something and export the G-code.
* _Expected Results_
  * When starting the second layer, the active extruder, i.e. whatever extruder was activated in the start G-code, heats up to the second-layer temperature. Slic3r inserts a G-code command like “M104 S235”. This causes the printer to change temperature on whatever extruder is currently active.
* _Actual Results_
  * The first extruder always heats up at layer 2, even when we're printing with the second one. Slic3r inserts an M104 command like “M104 S235 **T0**” when moving to layer 2. That T0 argument should not be there because Slic3r has no reason to assume we're using T0.
  * There are two possible workarounds:
    1. Explicitly assign the correct extruder in the object settings. This is cumbersome and I forget it in 50% of cases, which leads to failed prints and wasted time, because the issue only becomes obvious when the printer starts working on the second layer and the other extruder starts oozing all over the print, or marring top surfaces.
    2. Write a post-processing script that wipes the T0 argument when it detects single-extruder start G-code. This works for me, which is why I neglected to file this bug for a long time, but I still see people asking on Thingiverse why the wrong extruder heats up.

#### STL/Config (.ZIP) where problem occurs
[config.zip](https://github.com/alexrj/Slic3r/files/1048045/config.zip)

lordofhyphens, 2017-06-03:
https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/GCodeWriter.cpp#L86 There's the logic that is causing the behavior. 

Documentation for the Makerware and Sailfish behavior seems a bit thin on the ground, but since it doesn't apparently support the "Tx" syntax to change toolheads for later commands, I think the reason this works for you is a bug/undocumented feature.

Removing the two firmware checks would restrict it to just checking to see if multiple extruders are defined.

lordofhyphens, 2017-06-03:
@DrLex0 So removing the "T#" from M104 still gave you the correct behavior?

lordofhyphens, 2017-06-04:
Linked here from [/r/3DPrinting](https://www.reddit.com/r/3Dprinting/comments/6f59um/looking_for_information_about_sailfishmakerware/) to try to work out a way forward. 

@DrLex0 Another option with the overrides available is to add all of the extruder roles to your shortcuts and switch them on  your dual extrusion setup, although that wouldn't necessarily resolve the "which should I be using for my printbed" problem.

DrLex0, 2017-06-04:
@lordofhyphens In short, yes, if I edit the gcode file to omit the argument, the correct extruder is heated. However, I have been digging around in [Sailfish](https://github.com/jetty840/Sailfish-MightyBoardFirmware/blob/master/firmware/src/MightyBoard/Motherboard/Command.cc) and [GPX](https://github.com/markwal/GPX/blob/master/src/gpx/gpx.c) source code, and my conclusion is that the workaround of removing the T# argument only works thanks to **GPX**. It keeps track of the currently active extruder and appends the argument wherever necessary. If an M104 command has no T# argument, it just doesn't update its ‘gpx->current.extruder’ and writes out the binary command with the last set value.

This means the title of this bug is probably not the correct solution to this problem, because it seems awkward to rely on behavior of GPX, even though it may be the only thing in use to convert gcode to X3G for Sailfish.

Mind that there is no problem with the temperature values, also not for the printbed. Only one extruder is configured in my single-extruder profiles, therefore only one type of filament can be selected, therefore only the correct temperatures can be used. The problem is that Slic3r is unaware of what extruder is active at the moment it needs to insert the M104 command. It assumes that if only one extruder is configured, the extruder is always T0.

This is related to #3990 and maybe also #3976. Some ideas:
- Slic3r could inspect the start g-code and look for T0 and T1 commands to keep track of what extruder is active, a bit like how GPX does it.
- Decouple the extruder and tool number concept in the printer settings. Make the tool number an editable property for each extruder, and use that number for any T# arguments.