slic3r/Slic3r #4004: Brim select

Leozolt, 2017-06-02:
Hello.

thinking about a feature in the Layer tab. 
Selectable edges on the first layer with add / remove Brim functionality!

I am thinking about this because of #3993 I posted because an internal pillar doesn't get a brim.

Regards,

TT

lordofhyphens, 2017-06-03:
You are going to have to describe this in more detail, preferably with pictures.

Need to know what problems you are seeing and trying to solve.

Leozolt, 2017-06-04:
@lordofhyphens, Brim seems to only apply to outer shells(7). If I need to print a part with inner isles(1-6) I sometimes would like to have a brim on them(2,3,4,6) too.

I was thinking of beeing able to select the triangular isle(2) on the first layer and add a Brim.
Even select the hollow cylinder (4) and add a brim on the outside only (but still be able to add brim on the inside).

[image]

Hope this gives you more clarity.
[brim-example.zip](https://github.com/alexrj/Slic3r/files/1050704/brim-example.zip)

Regards, TT