slic3r/Slic3r #4005: Print Summary only shows up after you export GCODE

MidanRayam, 2017-06-03:
### Version
Slic3r-master.2017.06.02.1152.bcd8603.32bit

### Operating system type + version
Windows 10

### Behavior
The Print Summary panel on the sidebar which gives information about filament usage and cost shows up after you export GCODE. There is no reason for why it can't be shown after you sliced a model without exporting it as far as I know. So shouldn't it be shown when you select the preview tab and the slicing is finished?

lordofhyphens, 2017-06-03:
The information is not available until during a gcode export.

https://github.com/alexrj/Slic3r/pull/3669#issuecomment-275094923

If you want to do the refactoring necessary to make it available earlier then go knock yourself out, but it is enough work that I wouldn't touch it myself for so little benefit when there is so much else to do.

lordofhyphens, 2017-06-09:
@Jancomputerman @MidanRayam was talking about the new print statistics box, which @MidanRayam wanted to see before gcode export.

m-Th, 2017-07-06:
A simple fix can be done by exporting the G-Code in a temporary file / stream, taking the value from there and after this get rid of file / stream.