slic3r/Slic3r #4007: Z-Level Transformation Matrix Feature Request

emard, 2017-06-03:
Hotbed is not always completely flat surface,
This feature would allow reading a text file with stored XYZ points
in order to apply Z-level adjustment for uneven hotbed.

First layer Z should be adjusted to fit hotbed,
optionally subsequent layers could get linear
reduction of Z-correction until it  becomes zero, so 
top surface can be completely flat.

possible options to adjust z-level for first 4 layers
--z-level-adjust-file="z_level_calib.xyz"
--z-level-adjust-layers=4

z_level_calib.xyz
    # X Y Z
    0  0  0.1
    25 0 0
    50 0 0.1
    75 0 -0.1
    100 0 -0.1
    0 25 0
    25 25 0.1
    50 25 0.2
    ...

Z-level calibration file can be entered either manually by moving 
extruder and counting papers that slide below nozzle and entering XYZ in a
txt editor, or automaticaly from lucky owners of 3D printers which have hotbed 
level sensors and firmware support.

There exist z-autolevel g-code postprocessors for making PCB
with CNC like 
[Autoleveller](http://www.autoleveller.co.uk) or
[G-Code Ripper](http://www.scorchworks.com/Gcoderipper/gcoderipper.html)
Autoleveller it is not free and user must be extra careful to
review is postprocessed g-code is good for the
particular machine/firmware. Ripper doesn't work for extruder commands.

lordofhyphens, 2017-06-03:
This is already solved in most firmwares for 3D printing (our primary use case) and is also solvable with a gcode post-processing script. 

I won't say it has no place in Slic3r, but I have zero interest in adding it myself.

emard, 2017-06-04:
I understand it's a lot work. However if you or someone finds a suitable
post-processing script, it would be great to include it
in slic3r just to ease user trying many of them just to find
which one works with g-code output from slic3r