slic3r/Slic3r #4008: What is the logic behind "Arrange" operation?

cerdeira, 2017-06-04:
Version
Last Dev commit fccdd12

Operating system type + version
Win10 64 Home

 Behavior
After adding several parts to the plate, when you choose "Arrange", Slic3r spreds the parts all over the place even outside the defined plater.

[image]

lordofhyphens, 2017-06-04:
It's a general packing algorithm written, if I remember correctly, by @triffid . 
There's a note on a probably better algorithm in #3237 and we had an open GSoC project to port it to C++ so we could use it but nobody offered. 

#3998 #3452

This is probably a small regression -- Slic3r used to not do anything if it couldn't place everything on the bed. 

Also, optimal solutions are nontrivial (and generally NP-Hard).

triffid, 2017-06-05:
Yeah I wrote the initial auto pack algorithm, I know it's crude but I only
intended it to be a placeholder for a better one.. Is it still in there
even now?

cerdeira, 2017-06-07:
Auto arrange has a great potential.
It could be used to just laydown parts on plate, close toghether and aligned, or to move them around to minimise print time, or to group them by height to allow the user to aply mesch modifiers by height or layers, to group them by size or aerea, etc.