slic3r/Slic3r #4009: Crash on test/connect on Linux x64.

lordofhyphens, 2017-06-04:
1.3.0-dev 1d6170b

Observed behavior: Hard crash with coredump, segmentation fault. 

Backtrace:
````
#0  0x00007f71b6fbb26b in std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(std::string const&) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) back
#0  0x00007f71b6fbb26b in std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(std::string const&) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007f71b74ac3e0 in _List_node<std::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (this=<optimized out>)
    at /usr/include/c++/4.9/bits/stl_list.h:114
#2  construct<std::_List_node<std::basic_string<char> >, std::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (
    __p=<optimized out>, this=0x7f71a4a339d0) at /usr/include/c++/4.9/ext/new_allocator.h:120
#3  _M_create_node<std::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (this=0x7f71a4a339d0)
    at /usr/include/c++/4.9/bits/stl_list.h:514
#4  _M_insert<std::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (__position=..., this=0x7f71a4a339d0)
    at /usr/include/c++/4.9/bits/stl_list.h:1688
#5  emplace_back<std::basic_string<char, std::char_traits<char>, std::allocator<char> >&> (this=0x7f71a4a339d0)
    at /usr/include/c++/4.9/bits/stl_list.h:1039
#6  _M_initialize_dispatch<__gnu_cxx::__normal_iterator<std::basic_string<char>*, std::vector<std::basic_string<char> > > > (
    __last=<error reading variable: Cannot access memory at address 0xffffffffffffffe8>, 
    __first=<error reading variable: Cannot access memory at address 0xffffffffffffffe8>, this=0x7f71a4a339d0)
    at /usr/include/c++/4.9/bits/stl_list.h:1618
#7  list<__gnu_cxx::__normal_iterator<std::basic_string<char>*, std::vector<std::basic_string<char> > >, void> (__a=..., __last=..., 
    __first=..., this=0x7f71a4a339d0) at /usr/include/c++/4.9/bits/stl_list.h:640
#8  insert<__gnu_cxx::__normal_iterator<std::basic_string<char>*, std::vector<std::basic_string<char> > >, void> (this=0x541e150, 
    __last=..., __first=..., __position=...) at /usr/include/c++/4.9/bits/list.tcc:133
#9  Slic3r::GCodeSender::on_read (this=0x541dfb0, error=..., bytes_transferred=<optimized out>) at src/libslic3r/GCodeSender.cpp:371
#10 0x00007f71b74b50db in operator() (a2=<optimized out>, a1=..., p=<optimized out>, this=0x7f71a4a33cc0)
    at /usr/include/boost/bind/mem_fn_template.hpp:280
#11 operator()<boost::_mfi::mf2<void, Slic3r::GCodeSender, const boost::system::error_code&, long unsigned int>, boost::_bi::list2<const boost::system::error_code&, long unsigned int const&> > (a=<synthetic pointer>, f=..., this=0x7f71a4a33cd0)
    at /usr/include/boost/bind/bind.hpp:392
#12 operator()<boost::system::error_code, long unsigned int> (a2=<synthetic pointer>, a1=..., this=0x7f71a4a33cc0)
    at /usr/include/boost/bind/bind_template.hpp:102
#13 boost::asio::detail::read_until_delim_op<boost::asio::basic_serial_port<boost::asio::serial_port_service>, std::allocator<char>, boost::_bi::bind_t<void, boost::_mfi::mf2<void, Slic3r::GCodeSender, boost::system::error_code const&, unsigned long>, boost::_bi::list3<boost::_bi::value<Slic3r::GCodeSender*>, boost::arg<1> (*)(), boost::arg<2> (*)()> > >::operator() (this=this@entry=0x7f71a4a33ca0, ec=..., 
    bytes_transferred=<optimized out>, start=start@entry=0) at /usr/include/boost/asio/impl/read_until.hpp:433
#14 0x00007f71b74b555c in operator() (this=0x7f71a4a33ca0) at /usr/include/boost/asio/detail/bind_handler.hpp:127
#15 asio_handler_invoke<boost::asio::detail::binder2<boost::asio::detail::read_until_delim_op<boost::asio::basic_serial_port<>, std::allocator<char>, boost::_bi::bind_t<void, boost::_mfi::mf2<void, Slic3r::GCodeSender, const boost::system::error_code&, long unsigned int>, boost::_bi::list3<boost::_bi::value<Slic3r::GCodeSender*>, boost::arg<1> (*)(), boost::arg<2> (*)()> > >, boost::system::error_code, long unsigned int> > (function=...) at /usr/include/boost/asio/handler_invoke_hook.hpp:69
#16 invoke<boost::asio::detail::binder2<boost::asio::detail::read_until_delim_op<boost::asio::basic_serial_port<>, std::allocator<char>, boost::_bi::bind_t<void, boost::_mfi::mf2<void, Slic3r::GCodeSender, const boost::system::error_code&, long unsigned int>, boost::_bi::list3<boost::_bi::value<Slic3r::GCodeSender*>, boost::arg<1> (*)(), boost::arg<2> (*)()> > >, boost::system::error_code, long unsigned int>, boost::_bi::bind_t<void, boost::_mfi::mf2<void, Slic3r::GCodeSender, const boost::system::error_code&, long unsigned int>, boost::_bi::list3<boost::_bi::value<Slic3r::GCodeSender*>, boost::arg<1> (*)(), boost::arg<2> (*)()> > > (context=..., function=...)
    at /usr/include/boost/asio/detail/handler_invoke_helpers.hpp:37
#17 asio_handler_invoke<boost::asio::detail::binder2<boost::asio::detail::read_until_delim_op<boost::asio::basic_serial_port<>, std::allocator<char>, boost::_bi::bind_t<void, boost::_mfi::mf2<void, Slic3r::GCodeSender, const boost::system::error_code&, long unsigned int>, boost::_bi::list3<boost::_bi::value<Slic3r::GCodeSender*>, boost::arg<1> (*)(), boost::arg<2> (*)()> > >, boost::system::error_code, long unsigned int>, boost::asio::basic_serial_port<>, std::allocator<char>, boost::_bi::bind_t<void, boost::_mfi::mf2<void, Slic3r::GCodeSender, const boost::system::error_code&, long unsigned int>, boost::_bi::list3<boost::_bi::value<Slic3r::GCodeSender*>, boost::arg<1> (*)(), ---Type <return> to continue, or q <return> to quit---
boost::arg<2> (*)()> > > (this_handler=<optimized out>, function=...) at /usr/include/boost/asio/impl/read_until.hpp:480
#18 invoke<boost::asio::detail::binder2<boost::asio::detail::read_until_delim_op<boost::asio::basic_serial_port<>, std::allocator<char>, boost::_bi::bind_t<void, boost::_mfi::mf2<void, Slic3r::GCodeSender, const boost::system::error_code&, long unsigned int>, boost::_bi::list3<boost::_bi::value<Slic3r::GCodeSender*>, boost::arg<1> (*)(), boost::arg<2> (*)()> > >, boost::system::error_code, long unsigned int>, boost::asio::detail::read_until_delim_op<boost::asio::basic_serial_port<>, std::allocator<char>, boost::_bi::bind_t<void, boost::_mfi::mf2<void, Slic3r::GCodeSender, const boost::system::error_code&, long unsigned int>, boost::_bi::list3<boost::_bi::value<Slic3r::GCodeSender*>, boost::arg<1> (*)(), boost::arg<2> (*)()> > > > (context=..., function=...)
    at /usr/include/boost/asio/detail/handler_invoke_helpers.hpp:37
#19 boost::asio::detail::descriptor_read_op<boost::asio::mutable_buffers_1, boost::asio::detail::read_until_delim_op<boost::asio::basic_serial_port<boost::asio::serial_port_service>, std::allocator<char>, boost::_bi::bind_t<void, boost::_mfi::mf2<void, Slic3r::GCodeSender, boost::system::error_code const&, unsigned long>, boost::_bi::list3<boost::_bi::value<Slic3r::GCodeSender*>, boost::arg<1> (*)(), boost::arg<2> (*)()> > > >::do_complete (owner=0x5e6abe0, base=0x7f7198000c50) at /usr/include/boost/asio/detail/descriptor_read_op.hpp:104
#20 0x00007f71b74aeb9f in complete (bytes_transferred=0, ec=..., owner=..., this=<optimized out>)
    at /usr/include/boost/asio/detail/task_io_service_operation.hpp:38
#21 boost::asio::detail::epoll_reactor::descriptor_state::do_complete (owner=0x5e6abe0, base=0x5420120, ec=..., 
    bytes_transferred=<optimized out>) at /usr/include/boost/asio/detail/impl/epoll_reactor.ipp:651
#22 0x00007f71b74b1cc1 in complete (bytes_transferred=<optimized out>, ec=..., owner=..., this=<optimized out>)
    at /usr/include/boost/asio/detail/task_io_service_operation.hpp:38
#23 do_run_one (ec=..., this_thread=..., lock=..., this=0x5e6abe0) at /usr/include/boost/asio/detail/impl/task_io_service.ipp:384
#24 boost::asio::detail::task_io_service::run (this=0x5e6abe0, ec=...) at /usr/include/boost/asio/detail/impl/task_io_service.ipp:153
#25 0x00007f71b74b2186 in boost::asio::io_service::run (this=0x541dfb0) at /usr/include/boost/asio/impl/io_service.ipp:59
#26 0x00007f71b6ae9aea in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.55.0
#27 0x00007f71b9297064 in start_thread (arg=0x7f71a4a34700) at pthread_create.c:309
#28 0x00007f71b8fcc62d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
````

@alexrj if you'd like a copy of the coredump, I can send it to you.

lordofhyphens, 2017-06-04:
Oddly enough, it had been working fine earlier for some experimenting until Slic3r had hard crashed. 

I can connect with Printrun.

JanComputerman, 2017-06-09:
I also crash but not that hard, just shuts down slic3r, when trying to connect windows 8 64bit USB port com4 115k which is the usual port and speed for my PrintrBot Jr. 
HP Folio usb2.0 port Intel 5 CPU

JanComputerman, 2017-06-10:
Ok today the first time the PrintrBot Jr is powered up slic3r Controller connects and operates ...nice simple clean interface! But if disconnected will crash if connection is attempted, USB unplugged and in again crashes only a fresh restart of PrintrBot Jr allows the connection to be made. Com4 115k

lordofhyphens, 2017-06-22:
Looks like the bug is in the resend code according to this backtrace.

lordofhyphens, 2017-06-22:
I'm comfortable committing the minimal patch to avoid the crash, I will open up a separate issue to discuss what else can be done in this situation; indicating to the user to send M110 0 might be warranted, or doing it ourselves to get things into a known state (http://reprap.org/wiki/G-code#M110:_Set_Current_Line_Number)

closed 2017-06-22 by lordofhyphens

lordofhyphens, 2017-06-22:
@JanComputerman Host should be more stable now, you may need to send "M110 0" as a manual command occasionally.

commit b6708e87 (2017-06-22): Check that last_send buffer is large enough in addition to whether or not we have sent data.
Avoids a crash at reconnect, leaves an error message from printer (currently use M110 0 to get back to a good state)

Fixes #4009