slic3r/Slic3r #4010: PresetEditor in dialog mode incrementing/decrementing options by two

lordofhyphens, 2017-06-04:
### Version
1.3.0-dev 1d6170b

Linux x64
Linux rufus 3.16.0-4-amd64 #1 SMP Debian 3.16.39-1+deb8u2 (2017-03-07) x86_64 GNU/Linux
wxWidgets 3.0.2-1+b1 (from Debian packages)

Reproducible with http://dl.slic3r.org/dev/linux/slic3r-1.3.0-dev-1d6170b-linux-x64.tar.bz2

### Behavior
Slic3r GUI is incrementing/decrementing options by steps of 2 or more when using the mouse controls. Using up/down keyboard shortcuts are fine. 

### Reproduction Steps
1. Load any configuration. Even the default should do. 
1. In File->Preferences, set preset editor to dialogs 
1. Load a model
1. Adjust perimeters option via the up/down selectors with the mouse.
    * Option will change in steps of 2 or more. 

It looks to be some kind of latency/load problem having to do with processing the mousedown event and mouseup. The mouseup event seems to be delayed in processing long enough to indicate that the control was held down.

lordofhyphens, 2017-06-04:
This also could be related to the automatic shortcut creation and updating.

lordofhyphens, 2017-06-04:
If I have it in tab mode, incrementing/decrementing through the shortcut moves in steps of 1 for single clicks. The actual preset editor increments more than one. 

Same with dialog.

platsch, 2017-11-07:
Can't reproduce on plain Ubuntu:
OS: Linux 4.10.1-041001-generic x86_64 GNU/Linux, 
Wx: 3.0.2+dfsg-1.3ubuntu0.1
Slic3r: current master 1.3.0-dev 8ca6171a0a9e1723c774a609772691dcb4edf45f
click and scroll behavior works exactly as intended...