slic3r/Slic3r #4012: Rectangular Area center gap on top and bottom layers affected by infill Overlap setting.

JanComputerman, 2017-06-04:
### Version 
Problem in every version including the end of May 2017 included in Repetier
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions._
Sorry I am 15 miles from the computer running it ... but seems to be a long time problem going back 3 years and only recently associated it with the infill overlap setting.

_Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 8 

### Behavior
* _Describe the problem_
There is almost always a gap left in the center of a rectangular fill for the first and last (top) layers and the size was sometimes reduced by changing extruder width smaller or larger.

A secondary problem also occurred with a 15% infill Overlap setting ... I got bottom layers that were 1.2mm wide and 0.60mm thick using 0.6mm extrude and 0.30mm first layers on a spiral fill circular area. The nozzle then dragged itself through the first layer crossing to the other side of the object. This occurred repeatedly using 15% infill Overlap and was solved when set to 0%.

* _Steps needed to reproduce the problem_
Gap grows with increasing the % Overlap setting and gap can be decreased by going negative percent. It almost always occurs and I have resorted to adding a few lines of gcode to fill in the gap when needing a water tight print.

I will say that the infill Overlap setting greatly improves bridging when connecting to the vertical walls but there seems to be a gap problem when the walls are not parallel to the X or Y axis where the bridge doesn't reach the wall at all.

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2017-06-05:
RepetierHost's version of Slic3r is probably 1.2.9.

Please use 1.3.0-dev (http://dl.slic3r.org/dev/win) and check to see if the
problem still exists.

lordofhyphens, 2017-06-05:
When you are able, please add the required information  (sample 3d model, exported config.ini, etc).

We cannot reproduce without that information. If we cannot reproduce it, we cannot fix it.

JanComputerman, 2017-06-05:
I'm using 1.2.9 slic3r which I updated last month after using a 3 year older version. Windows PC I use stays off line so win updates will not crash long prints which it was doing monthly installing updates I didn't need.
Looks like it still is far away from angled edges ...photo attached ... 0% then when I used 20% it moved closer but still will drop filament once in a while.
Will get a slic3r update when I get better wifi access and test it further.

Thanks for getting back! Geez even cell service is poor tonight.

Jan

--------------------------------------------
On Mon, 6/5/17, Joseph Lenox <notifications@github.com> wrote:

 Subject: Re: [alexrj/Slic3r] Rectangular Area center gap on top and bottom layers affected by infill Overlap setting. (#4012)
 To: "alexrj/Slic3r" <Slic3r@noreply.github.com>
 Cc: "JanComputerman" <lookinnwpt@yahoo.com>, "Author" <author@noreply.github.com>
 Date: Monday, June 5, 2017, 12:04 AM
 
 When you are able, please add the
 required information  (sample 3d model, exported config.ini,
 etc).
 We cannot reproduce without that information. If we
 cannot reproduce it, we cannot fix it.
 
 —
 You are
 receiving this because you authored the thread.
 Reply
 to this email directly, view
 it on GitHub, or mute
 the thread.

JanComputerman, 2017-06-06:
See photo ... managed to download Slic3r on my phone unzip it and ran it on the part ...I could send you the openSCAD file it's not long. Interesting that the bottom layer looks closer to the side wall. Are you calculating the wall from the wrong Z height?
--------------------------------------------
On Mon, 6/5/17, Joseph Lenox <notifications@github.com> wrote:

 Subject: Re: [alexrj/Slic3r] Rectangular Area center gap on top and bottom layers affected by infill Overlap setting. (#4012)
 To: "alexrj/Slic3r" <Slic3r@noreply.github.com>
 Cc: "JanComputerman" <lookinnwpt@yahoo.com>, "Author" <author@noreply.github.com>
 Date: Monday, June 5, 2017, 12:04 AM
 
 When you are able, please add the
 required information  (sample 3d model, exported config.ini,
 etc).
 We cannot reproduce without that information. If we
 cannot reproduce it, we cannot fix it.
 
 —
 You are
 receiving this because you authored the thread.
 Reply
 to this email directly, view
 it on GitHub, or mute
 the thread.

JanComputerman, 2017-06-06:
//OpenSCAD partial of object where the gap condition occurs 
// yeah its pretty slick use of the commands to make interesting shapes
$fn=8;
D=10.71;
hull(){
translate([0,0,0]{$fn=60;cylinder(d=15,h=10);} // outer round end
translate([15,0,D/2-0.4]){rotate([22.5,0,0]){rotate([0,-90,0]){cylinder(d=D,h=0.1);}}} // 8 sided shaft
}

JanComputerman, 2017-06-06:
As for slicing setting i use 0.5mm extrude with 0.40 nozzle, e3dv6 extruder, 0.25 0.25 layers, 3 top 3 bottom, no fill, 2-3 perimeters 0-20% infill Overlap doesn't matter it doesn't reach the wall on the initial 3rd from the top layer, it does better 2nd from the top and then the top leaves a gap in the center of the fill surface at least 0.50mm wide in most cases.

JanComputerman, 2017-06-06:
[image]
[image]

JanComputerman, 2017-06-07:
[image]

Screen shot of layer in question from latest Slic3r version...

lordofhyphens, 2017-06-07:
... alt-printscreen will copy the active window to the clipboard. You can then paste here.

You should also be posting the 3d preview as that actually tells you how wide the traces are

lordofhyphens, 2017-06-07:
Unless you are running something particularly weird like some particularly strange remote terminal situation.

lordofhyphens, 2017-06-07:
Run file->Export Config in Slic3r. Zip that file up and upload it here.

JanComputerman, 2017-06-08:
The computer I print with does not go online otherwise windows forces updates into it that I don't want and then reboots at random times even when set to do it at night. I basically web surf with an iPad which as you know has limited connectivity to a PC or anything else for that matter ...that's another story with iTunes wiping out podcasts and uncontrollable backups. I just take a photo of the screen with my phone camera and email it. I did update GitHub with the photos of the preview shot with the cursor pointing to the problem area in the 3rd from the top layer ... That zigzag just fell inside the print but I managed to prop it up with a toothpick so the next layer pressed it back down and bonded to it. Also entered some of the openSCAD code that generated that part so you can generate it and slice it. Just so you know I also have printed cubes 40mm across bridging and got fairly flat tops so it's unique to this angled sides that has the issue with not reaching the outer vertical walls.
--------------------------------------------
On Wed, 6/7/17, Joseph Lenox <notifications@github.com> wrote:

 Subject: Re: [alexrj/Slic3r] Rectangular Area center gap on top and bottom layers affected by infill Overlap setting. (#4012)
 To: "alexrj/Slic3r" <Slic3r@noreply.github.com>
 Cc: "JanComputerman" <lookinnwpt@yahoo.com>, "Author" <author@noreply.github.com>
 Date: Wednesday, June 7, 2017, 5:36 PM
 
 ... alt-printscreen will copy the
 active window to the clipboard. You can then paste here.
 You should also be posting the 3d preview as that
 actually tells you how wide the traces are
 
 —
 You are
 receiving this because you authored the thread.
 Reply
 to this email directly, view
 it on GitHub, or mute
 the thread.

JanComputerman, 2017-06-08:
Well that was a waste of 10 Meg's of my 30megs a month phone plan ... I tried to select the zip file but it doesn't show zip files when I look in the folder I put the file into. Grrr downloads is not the download folder ok I got it now ... 
[slic3r_settings.zip](https://github.com/alexrj/Slic3r/files/1059717/slic3r_settings.zip)

lordofhyphens, 2017-06-08:
For anyone else trying to figure out what is going on here (and not in a mood to debug OpenSCAD syntax errors):
[4012.zip](https://github.com/alexrj/Slic3r/files/1059803/4012.zip)

lordofhyphens, 2017-06-08:
Negative overlap for infill/perimeters does what I think it would do: move the outer ring of infill in (away from the perimeters), but only for the internal infill. Top/Bottom solid infill is indeed not affected by the overlap setting. This may be by design; @alexrj would know more. 

[image]

The "gaps" in the top solid infill look like generally unavoidable holes in the pattern--the nozzle isn't small enough to put a loop there. We don't apply gap fill to top solid infill, and finding a "perfect" path to completely cover (and including changing width in there) is **very** expensive in terms of computation time. 

FYI, your extrusion width of 0.25mm on a 0.4mm nozzle is probably going to cause problems for you. 

Infill overlap's default value is now 55% by the way.

JanComputerman, 2017-06-08:
The 0.25 width is for the infill which produces very fine extrudes with typically 5-10% fill which is good enough when I use it but this has 0% fill. Top and bottom layers are set to 0.50mm width which is what is not quite working and the overlap will move everything inwards except the outer most perimeter if negative and outwards if positive but still leave a big enough gap between the bridge and inner wall to not connect to it in the angled areas. I understand that you can't get every little gap in the surface filled in but this is not the problem with this bridging issue ... It actually pulls away from the angles sides and is closer and contacting the circular and straight sections on either side of it. That preview is new to me ... Never seen it before ...looks great. I run Repetier host with slic3r imbedded so I still figuring out the Slic3r console.

lordofhyphens, 2017-06-08:
That is the 3d preview on Slic3r. You can use it as a host too, no need for
RepetierHost for a lot of use cases.

You seemed to have 3 different issues going on so I split them.

JanComputerman, 2017-06-08:
Yeah in Repetier it actually views the gcode and that was having the bridging issue which when viewed in slic3r seems to not exist ... I think when my brother is out of the house I will print it again and see what happens. I haven't even mentioned the first layer has to be 0.20mm limit either so yeah separate issues but related to the overlap to some extent. As for the gap at the center of the top and bottom layers yes a single extrude fills it so if you must have room for two passes that is the reason it leaves that gap. I will also say that I did some testing with adding some z axis motion when bridging to compensate for the filament sag as it extruded but ultimately the real trick is knowing the volume to extrude for the distance to bridge with a nozzle size diameter of filament and extrude about 10% less to stretch it tight.

JanComputerman, 2017-06-08:
My main reason for not upgrading Repetier was the extensive numbers of configurations I had created for all the different filaments I use, nozzle sizes and print settings. It was after some research on where the settings were stored and backing them up did I upgrade and now Slic3r console settings is finding only some of them ... It will take some time to get used to it.

lordofhyphens, 2017-06-08:
With the old version of Slic3r, export a config bundle. Import it with the
new version.

Ensure that everything has saved. Enjoy the new features.

JanComputerman, 2017-06-08:
I will thanks! Hopefully there is a nice printer interface as well once I configure the USB port... I will admit that I occasionally use Cura for objects with round tops but not much else unless there are weird cgode paths generated by Slic3r ... Internal verticals or very narrow vertical gaps that need to be clean sometimes work very well with Cura.

JanComputerman, 2017-06-09:
Back to the center gap of a concentric fill first layer ... A rectangular area first layer ends up with a gap at the center with a first layer 0.30mm BUT changing it to 0.31mm eliminated the gap! Several other first layer thicknesses still generated a gap. Weird, but I can work with this "feature".