slic3r/Slic3r #4014: Overhang threshold not working correctly.

RichyT, 2017-06-06:
### Version
1.2.9
### Operating system type + version
Windows 7

### Behavior
Overhang threshold is not behaving as expected. I have a hollow dome and the threshold overhang setting is 45 degrees. Support appears to only be being generated for angles >60 degrees.

I have created a simpler file which illustrates the issue with rotating blocks. Overhang setting is 45 degrees but support is not generated until 65 degrees.

I changed the setting to 80 degrees and support was generated in the correct place but no support was generated for a block at 75 degrees.

With a setting of 60, support was generated at 40 degrees but not 45

A setting of 65 shows support at 45 (but not at 50 and above)

#### STL/Config (.ZIP) where problem occurs

lordofhyphens, 2017-06-06:
Use 1.3.0-dev, please (dl.slic3r.org/dev) and please note that the semantics for that option has changed to also include % of perimeter width (old default was 200%).

RichyT, 2017-06-06:
Have done. Issue persists. Have updated issue report and attaching screenshots.

[image]
[image]

RichyT, 2017-06-06:
Note that if I set 50%, I appear to get correct support. The issue appears to be only for the angle.

Martronic-SA, 2017-06-06:
Be careful, 45° = 100% not 50% right?

lordofhyphens, 2017-06-06:
50% is relative to the extrusion width. If your extrusion width and height
are the same then it would be ~45°.

Martronic-SA, 2017-06-06:
Ok, I was sure it was a slope value expressed in %! My bad.

Martronic-SA, 2017-06-06:
But if the width is the same as height, it should be 100% no?

RichyT, 2017-06-07:
I took it as it meant that the perimeter moved over 50%. But that would only be valid if the height was the same as the width. 45 degrees should always be 45 degrees though.

(And yes, it was definitely 45 and not 45%)

RichyT, 2018-12-18:
I just downloaded the latest Repetier Host with 1.3.1-dev and it appears to be working correctly now. Nice job, whoever fixed it!

closed 2018-12-18 by RichyT